lgsvl/simulator #978: Velodyne Lidar Sensor Source Code

arslansarp, 2020-09-12:
Hi, I am trying to find velodyne lidar 32 channel source code with udp port. However, I couldn't find the source code. I have got only build version.

I am trying to access it to look how udp works, and other stuff. Since rosbridge so exhausted  after adding so many sensors. I am considering sending other sensor datas over udp namely camera. 

Could you upload the velodyne lidar sensor 32 channel with udp source code please ?

EricBoiseLGSVL, 2020-09-14:
@arslansarp We do not supply the source code for the UPD Velodyne LiDAR sensor at this time.  We are looking to release sometime later this year.

rudolfkrecht, 2021-03-12:
Hi, could you please provide an update regarding the availability of the source code for UDP lidars, or is there any blank script for UDP lidars? We would like to simulate 64 channel lidars.
Thank you in advance.

EricBoiseLGSVL, 2021-03-13:
@rudolfkrecht We are discussing internally if we are able to release source on this.  As of today, we do not.