lgsvl/simulator #979: how to load assetbundle map to unity

qwetqwe, 2020-09-13:
Hi team,
I want to annotate BorregasAve map in unity. But I did not find it unity project in repositories. So I want to load asset bundle of the map.
So I learn the way from internet:
1. create an object
2. attach the object to a load script in inspector
3. the script is created as: 
```
using System.Collections;
using System.Collections.Generic;
using UnityEngine;
using System.IO;
public class assetbundles : MonoBehaviour
{
    // Start is called before the first frame update
    void Start()
    {
        var myLoadedAssetBundle 
            = AssetBundle.LoadFromFile("Assets/External/Environments/environment_BorregasAve/c4c227f1-efc6-4036-b64f-e68203b4c368_environment_main_windows");
        if (myLoadedAssetBundle == null) {
            Debug.Log("Failed to load AssetBundle!");
            return;
        }
        var prefab = myLoadedAssetBundle.LoadAsset<GameObject>("MyObject");
        
        Instantiate(prefab);
    }

    // Update is called once per frame
    void Update()
    {
        
    }
}
```
but it got error 
```
invalidOperationException: This method cannot be used on a streamed scene AssetBundle.
UnityEngine.AssetBundle.LoadAsset (System.String name, System.Type type) (at <9eadb6c7050a4e92b206c83cb4765a54>:0)
UnityEngine.AssetBundle.LoadAsset[T] (System.String name) (at <9eadb6c7050a4e92b206c83cb4765a54>:0)
assetbundles.Start () (at Assets/assetbundles.cs:16)
```
I'm fresh of unity and c sharp. Is there any way to load asset bundles?
Thanks a lot!

EricBoiseLGSVL, 2020-09-15:
@qwetqwe You cannot edit the annotation BorregasAve map, sorry.  This is not a map that we released for open source, so you cannot load in Editor to edit. To create your own you can:
- Purchase assets in the Unity Asset Store and create a map (We have tutorials in our documentation)
- Import HDmap and use our tools to create a mesh from the annotation
- Contact us for digital twin services to create a map
What exactly are you trying to do with the annotation change?

qwetqwe, 2020-09-15:
@EricBoiseLGSVL 
Hi,Eric. I want to test parking function in apollo. I see that only BorregasAve has parking lot. But its HDmap does not. So I want to annotate its parking space polygon and guide a lane to parking space in apollo HDmap

EricBoiseLGSVL, 2020-09-15:
We have open source San Francisco that has parking lots you can annotate and create bundles from source.  You can also use AutonomouStuff map that has parking lots but this is not open source.

EricBoiseLGSVL, 2020-09-15:
Spot in SF that would need annotated:
[image]

AutonomouStuff is already annotated for this
[image]

closed 2020-09-21 by qwetqwe