lgsvl/simulator #980: How to create a custome vehicle model

coco721, 2020-09-14:
We would like to lgsvl simulator as the platform to verify our algorithm.Howerver the vehicle that we will be using is tractor in airport. So we need to modify the current model or even create new one with 3D-modeling software. Is there any instruction on how to do that,thx.

momala, 2020-09-14:
Hello, I am also doing sth like you.
I think this video would be useful 
https://www.youtube.com/watch?v=R_Z65kimCII&ab_channel=LGSVLSimulator
for importing the cad model I think working with the Blender can help a lot.

rongguodong, 2020-09-14:
Please check this instruction:
https://www.lgsvlsimulator.com/docs/add-new-ego-vehicle/

EricBoiseLGSVL, 2020-09-15:
@coco721 Be careful with the hitch connection and unity physics. It's not easy to work with.  I recommend looking at hinge and fixed joint tutorials to get this right.