lgsvl/simulator #981: Lidar sensor detection and mesh collider does not meet

momala, 2020-09-14:
Hi,

I made a new vehicle from scratch and import it inside the simulator. it seems everything is fine unless the Lidar sensor detection that somehow does not meet the collider geometry. 
As you can see in the picture the lidar is mounted at the top back of the vehicle. I use a simple cube as a collider similar to the vehicle size. But what is strange for me is the points that detected in front of the vehicle which means that the lidar passes from the collider and also worse that is the points that are detected in the sides on the body. The lidar is mounted in the central plane. 

[image]
[image]

Is there anything wrong with our collider or vehicle model?

rongguodong, 2020-09-14:
Looks like there is something invisible/transparent visible to the camera of LidarSensor.
Why do you use a collider, not the vehicle itself, for point generation? Can you check the size of the collider? How does it look like?

EricBoiseLGSVL, 2020-09-15:
@momala Nice vehicle!  Why not use a MeshCollider set to Convex?  This would give a more accurate collision.  This also might be collision happening from inside the mesh.  Do you have normals right and or faces inside the mesh?  If so, you can put these and other mesh geometry on a physics layer that is ignored by LiDAR raycast.

momala, 2020-09-22:
> Looks like there is something invisible/transparent visible to the camera of LidarSensor.
> Why do you use a collider, not the vehicle itself, for point generation? Can you check the size of the collider? How does it look like?
I converted this model from SolidWorks by using blender as a converter. first I used sheet metal for the body, and I thought maybe it was the problem because it had not any thickness. I thickened the body and give a small thickness to it and it also did not work.
here is my collider: it is almost similar to the body of the vehicle.
[image]

> @momala Nice vehicle! Why not use a MeshCollider set to Convex? This would give a more accurate collision. This also might be collision happening from inside the mesh. Do you have normals right and [...]
Thank you very much. it is our autonomous shuttle. 
if you mean the convex in the mesh collider setting, I should say that it was already checked.
[image]
Inside my vehicle is empty and this is inside it in lgsvl simulator:
[image]

Again I put the lidar in front of the vehicle, Also I don't know why I can't change the lidar TF rotations, everything I input as the roll pitch yaw it doesn't work. 
[image]
[image]
[image]

It is so strange for me why it detects the sides and also the other side windshield?
I want to put some rigid blocks inside the vehicle and see what happens!!
I read somewhere in this forum that the lidar sensor does not use the collider, so what's your idea? I appreciate it if you can help me to solve this problem.

EricBoiseLGSVL, 2020-09-22:
@momala Sorry, it's not the collider that gets hit with the lidar but the mesh.  You can set the lidar to ignore a mesh if you set the vehicle Physics Layer to Sensor, but it will pass right through.  It looks like you have the lidar set at a radius from the lidar sensor center in a wide circumference so the beam is starting inside the vehicle.  I would change the body mesh layer to Sensor and also look at the lidar settings.  You could also remove geometry from inside the vehicle so normals are facing out only and no mesh collision will happen.

martins-mozeiko, 2020-09-22:
This happens due to near plane of lidar camera - it clips off whatever is close-by it, that's why near parts of the vehicle does not block lidar rays, and they hit vehicle from "inside" on opposite side.

Solution is either to move lidar a bit away from vehicle. Or decrease near plane distance for lidar camera - but be careful about decreasing it too much, too small values will greatly reduce lidar rendering precision.

momala, 2020-09-28:
> This happens due to near plane of lidar camera - it clips off whatever is close-by it, that's why near parts of the vehicle does not block lidar rays, and they hit vehicle from "inside" on opposite [...]

> @momala Sorry, it's not the collider that gets hit with the lidar but the mesh. You can set the lidar to ignore a mesh if you set the vehicle Physics Layer to Sensor, but it will pass right through. [...]

Thank you very much for mentioning that. Actually, you are right, As I make it farther, it can detect the windshield.
[image]

Here is my setting for the lidar:
```
{"type": "Lidar", "name": "Lidar",
            "params": {"LaserCount": 32, "MinDistance": 0.5, "MaxDistance": 100, "RotationFrequency": 5, "MeasurementsPerRotation": 360, "FieldOfView": 41.33, "CenterAngle": 15, "Compensated": true, "PointColor": "#ff000000", "Topic": "/points_raw", "Frame": "velodyne"},
            "transform": {"x": 0, "y": 1.5, "z": -1.1, "pitch": 0, "yaw": 0, "roll": 0.5}},
```
But I have another problem that I cant rotate the frame of the lidar. I change the roll pitch yaw but nothing happened. do you have any idea?

luqiang21, 2020-09-28:
@momala Rotations are in degrees, did you try with degree numbers?

momala, 2020-09-29:
> @momala Rotations are in degrees, did you try with degree numbers?

Thank you very much. I thought they were in radian. Now by decreasing the MinDistance to 0.2 my problem almost solved. Thank you very much for your supports.