lgsvl/simulator #982: [Suggestion] add a function to pick coordinate point in simulator

qwetqwe, 2020-09-15:
Hi team
I think picking coordinate point in map is really important. because we need absoulute coordinate to add NPCs and annotate map.It seems that lgsvl does not have this function. lgsvl can only get point by mouse in unity, but that is not convenient.Could team add a function to get point in simulator just like dreamview:
[image]
Thanks a lot

EricBoiseLGSVL, 2020-09-15:
@qwetqwe This could be a good idea.  Can you describe a bit more about the feature?
- We do have sensors that output lat/long coordinates, what UX/UI change to make this better/easier to use?
- Is it just coordinates that match dreamview? How would this work for other AD stacks?

qwetqwe, 2020-09-15:
@EricBoiseLGSVL 
We can add a button in the menu or keyboard shortcut. After pressed, the mouse pointer can select point in topview in simulator. Rightdown will show the point coordinate.UTM will be better. We can also select a list of points. the points' coordinates can be copied or export a file.
That is not related with dreamview. It is just what I want.

qwetqwe, 2020-09-15:
like this:
[image]

lemketron, 2020-09-15:
Like this? Just "visualize" the GPS Odometry sensor...
[image]

qwetqwe, 2020-09-15:
Yes, you have to "drive"  the car to the place and it is not accurate. I think it will be better to get point directly by mouse. 
I'm sorry that I can't see the youtube video. Could you describe it?

EricBoiseLGSVL, 2020-09-15:
@qwetqwe The video is showing our Visual Scenario Editor, where you can set NPC waypoints and replay the scenario simulation.  This has a top down view, with mouse clicked waypoints.  Is this what you are trying to do with the coordinate system?

qwetqwe, 2020-09-16:
@EricBoiseLGSVL , 
It seems that is what I want. Is it in unity or simulator?

EricBoiseLGSVL, 2020-09-16:
@qwetqwe It is not officially released yet.  We are putting the final touches and should be done soon, sorry.  In the meantime you can write a class that takes mouse position, raycasts to ground to get positions that you want while in free cam mode.  This can be written to a UI element you add and or output to a file.

closed 2020-09-21 by qwetqwe