lgsvl/simulator #983: Export HD map issue?ArgumentNullException: Value cannot be null. Parameter name: key

Addi2020, 2020-09-15:
ArgumentNullException: Value cannot be null.
Parameter name: key
[image]

EricBoiseLGSVL, 2020-09-15:
@Addi2020 You are missing lane data to export.  Please look at our open source map, CubeTown, to see simple annotation and what is required for HDMap export.  @luqiang21 can confirm

luqiang21, 2020-09-15:
@Addi2020 Are you using our latest release? Each lane needs boundary lines annotated as well. I just tried with 2020.06 release with only two new lanes, it is throwing correct error:
`There are no boundary lines for some lanes! please annotate boundary lines for each lane OR use "Create lines" button in HD Map Annotation window to generate boundary lines!`

Addi2020, 2020-09-15:
@luqiang21 , I am using 2019.3, because my project works on this release. if I use the latest version, it gives me a lot of errors.
Even i create boundry lanes , it still give an error, but i will try again. 

I am working on creating HD map for a Self-driving car (Baidu Apollo5.5 Project). 
I follow the folwing steps

1. Import .pcd file
2. annotate lanes , boundries, junctions, cross walk etc
3. export HD map

I follow this document  https://www.lgsvlsimulator.com/docs/map-annotation/.

If there is any other document or video related to HD map annotation using Unity please shear here.
Also, share the link of (cube town map).

luqiang21, 2020-09-15:
@Addi2020 I see, once you create boundary lines, make sure to link them correctly with each lane.
Here is the link to cube town: https://github.com/lgsvl/CubeTown

EricBoiseLGSVL, 2020-09-15:
@Addi2020 We highly recommend you update to the latest simulator version if you are working with Apollo5.5.  We have many fixes and improvements.  Please post issues updating here and we can help you fix them.

Addi2020, 2020-09-16:
Thanks @luqiang21 
@EricBoiseLGSVL 
I use the latest release 2020.1.5. when i use it, My project shows a lot of errors, and also the simulator tab is not available. So I cannot import .pcd file or work on HD map annotation.
[image]

luqiang21, 2020-09-16:
Hi @Addi2020, the latest release version is 2020.06, did you get the correct version?

EricBoiseLGSVL, 2020-09-16:
Also, your Unity version is not right.  2019.3.15f1.  Our release is 2020.06.  It is a branch.  Please see the release [here](https://github.com/lgsvl/simulator/releases/tag/2020.06)
- Close Unity
- Update Unity to 2019.3.15f1 from download archive.  Be sure to have Linux and Windows build modules
- Pull latest 06 release
- Delete Library folder in the simulator repo root
- Open Unity, let it make the Library folder again
- Close Unity
- Open Unity and most errors should be fixed with Unity.

Addi2020, 2020-09-17:
@EricBoiseLGSVL  thanks for the detail description. I will try it now.

@luqiang21 thanks for the suggestion , the map extraction problem is solved by adding boundary lines and link them with corresponding lanes.

closed 2020-09-18 by Addi2020

AlexandrZabolotny, 2020-12-14:
I have a similar problem.  
I added boundary lines and linked them to the corresponding stripes.  
But a similar error occurs.   

[image]  

Tell me how you can understand where this error occurs on the map?

EricBoiseLGSVL, 2020-12-14:
@AlexandrZabolotny Uncheck is Self reverse lane and display lane.  Then be sure boundry lines are not child objects of the map lane.
You also have too many MapLanes in one LaneSection.  They are supposed to be per maplane area, not a overall area for all maplanes.
[image]