lgsvl/simulator #984: Python API to put obstacle in a parking slot.

Maddy2927, 2020-09-15:
Hey LGSVL team, i need to write a pythonAPI wherein i can place few obstacle(pedestrian,cone) in a map and then move my pedestrian in a certain path when my ego car comes close to it. 
Please let me help with the script.
Thanks in advance.

EricBoiseLGSVL, 2020-09-15:
@Maddy2927 You can create waypoints for the pedestrian to walk and add the traffic cone controllable.  As for triggers to move the pedestrian when the EGO is close, you will need to modify simulator code or use the Visual Scenario Editor to create a scenario.  This is still in development and only works with Linux because of a required runner to convert scenarios to API classes.

Maddy2927, 2020-09-16:
Also, @EricBoiseLGSVL i wanted to know whether for adding traffic cone in the map, is using unity necessary or directly we can change the pythonAPI and produce the cone in the map like vehicle and pedestrian.
I tried to add the traffic cone assetbundle and run the already given cone API, but was getting an error and cone was not getting spawned.

EricBoiseLGSVL, 2020-09-16:
@Maddy2927 Unity Editor is not needed, just easier to place static objects, esp if new objects are needed.  Traffic Cone controllable should be working with the quick script, we might have missed a commit in the release.  Can you please post error log and any images that will help debug?

Maddy2927, 2020-09-22:
@EricBoiseLGSVL Also, you mentioned trigger function to move the pedestrian when ego vehicle is close. Can this trigger function be implemented in the older version 2020.05 version as well? Or 2020.06 is necessary for it?

EricBoiseLGSVL, 2020-09-22:
@Maddy2927 The VSE is still in development, sorry.  We will release this soon.  You can make a controllable that has trigger components in 05 or in 06 release.  This is a Unity API component and it doesn't matter which one.

With the new controllable class, you can create OnTriggerEnter method to trigger when an object.tag == Player happens and run any logic you want.  Please see the RadarSensor to see triggers working.

So you create a controllable plugin.  Add a trigger volume to the controllable, set the size.  Add the YourNewControllable.cs class that implements the controllable interfaces, write an OnTriggerEnter method.  Check for Player in the method.  If true, use the logic you provide.

marcusvinicius178, 2021-04-07:
Hi @EricBoiseLGSVL you have mentioned in your first answer that the VSE just work on Linux properly. I tried to use it along AutowareAi and failed. I ask myself if failed because I am using AutowareAi from a docker image? Should I install AutowareAI from source and then VSE would work properly with AutowareAI bridge?

lemketron, 2021-04-07:
VSE is now available in 2021.1 simulator. There are many other fixes as well in the simulator and the latest Python API.

closed 2021-04-07 by EricBoiseLGSVL