lgsvl/simulator #986: Maximum nu of multi agents that can be supported

neeraj310, 2020-09-15:
Hi Support Team

I am working on a survey for LGSVL capabilities and need few clarifications regarding the number of supported agents.

a) What is the maximum number of agents that can be supported on a normal desktop(16GB RAM, GPU)

b) Is it possible to add custom vehicles. I want to design my own vehicle and use it inside simulation framework.

Neeraj

EricBoiseLGSVL, 2020-09-15:
@neeraj310 Welcome to LGSVL Simulator
a) EGO and or NPCs/Pedestrians? What is the GPU that you are using?  Linux or Windows?  What AD stack do you plan on using?
b) Yes, we have tutorials on how to do this in our documentation for EGO vehicles.  NPCs are a little different but are easy to copy existing and add your own.

neeraj310, 2020-09-17:
Hi Eric 

Thanks for the quick response. 
I am planning to use Linux, GEFORCE GTX 1080 GPU. As I am working on a survey, I am interested in finding the max limit of agents that can be supported simultaneously.

Neeraj

EricBoiseLGSVL, 2020-09-17:
@neeraj310 Max dynamic objects in the scene can vary depending on the sensors/bridge/polygon count/textures/materials/shaders/alpha/GPU/CPU.  1080 is an older GPU and won't be able to push the limits of Simulator or HDRP.  
- Currently, we cap NPC's to max 30 and one EGO with simple sensors for lower end of the spectrum hardware in BorregasAve map with a useful FPS on a single local simulation.  This is NOT running AD stack on the same machine in tandem.
- You can make simple box NPCs, EGO and a low poly map to increase agent amounts.
- With our API you can run slower than real time and increase this max number.  
- Lastly, you can use our distributed simulation feature that is only limited by the hardware you have.  A simulation can run on N machines and the EGOs/NPCs/Data CPU/GPU load can mitigated. 

It really depends on what you are trying to do and assets you plan to use.  Hope this helps