lgsvl/simulator #987: Question about import point Cloud file

wisdom-bob, 2020-09-16:
Recently, I'm trying to annotation map according to pointCloud. Now, I only realise to import pointcloud with intensity feature. I hope you can give a .pcd file that can be successufully imported with color feature!

EricBoiseLGSVL, 2020-09-16:
@kamkolak Did the RGB field in PCD file fix get released in 06?  I think this is a referencing #921

wisdom-bob, 2020-09-17:
The problem is that most .pcd file support PCL function library. However, this tool(lgsvl-simulator) does not support .pcd / .ply files that meet PCL function library requirements, whether there is color filed or not. This is the reason why I hope you can provide me with a point cloud file with color fileds, and I can parse it by myself.

wisdom-bob, 2020-09-17:
> The problem is that most .pcd file support PCL function library. However, this tool(lgsvl-simulator) does not support .pcd / .ply files that meet PCL function library requirements, whether there is [...]

or if you can give me some support how to create point cloud file the meets requirements of the lgsvl-simulator, I will be very very thanksful.

EricBoiseLGSVL, 2020-09-18:
Adding @daviduhm on this since I can't assign @kamkolak

EricBoiseLGSVL, 2020-09-18:
@wisdom-bob We will need to discuss internally when we can add full PCL support to our roadmap.

martins-mozeiko, 2020-09-18:
PCL is just a library that loads, processes and outputs point cloud files. The format of files it supports can be almost anything. It supports loading & writing any format.

But the simulator requires specific fields to be set in point cloud for it to make sense - like xyz for position, color/intensity for coloring. But it won't use other attributes - pcd file can have any amount of extra info associated to each point.

To use pcd file with simulator, you need to prepare it to have this necessary information in pcd file. If you have custom data in there, then either you convert it to whatever simulator supports (xyz/intensity), or modify C# import code to use your custom data. Simulator cannot automatically support anything that pcd contains - it's a custom logic that you need to implement somewhere.

Here's code code for pcd import: (PcdPointProcessor.cs)[https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Editor/PointCloud/Trees/PcdPointProcessor.cs]. There's also Ply, Las and Laz format importers in same folder.

For example, here's an pull request that implements RGB color import from pcd files: https://github.com/lgsvl/simulator/pull/770

wisdom-bob, 2020-09-19:
> PCL只是一个加载，处理和输出点云文件的库。它支持的文件格式几乎可以是任何格式。它支持加载和写入任何格式。
> 
> 但是模拟器需要在点云中设置特定的字段才能使其有意义-例如xyz表示位置，颜色/强度表示着色。但是它不会使用其他属性-pcd文件可以具有与每个点关联的任意数量的额外信息。
> 
> 要将pcd文件与模拟器一起使用，您需要准备使其在pcd文件中具有此必要信息。如果您有自定义数据，则可以将其转换为模拟器支持的任何格式（xyz / intensity），或修改C＃导入代码以使用您的自定义数据。Simulator无法自动支持pcd包含的任何内容-这是您需要在某个地方实现的自定义逻辑。
> 
> 这是pcd导入的代码：[PcdPointProcessor.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Editor/PointCloud/Trees/PcdPointProcessor.cs)。同一文件夹中还有Ply，Las和Laz格式的导入器。
> 
> 例如，这是一个未合并的拉取请求，可实现从pcd文件导入RGB颜色：[＃770](https://github.com/lgsvl/simulator/pull/770)

just as you says, I know the simulator can not support all pcl format. I will try the #770. I hope this will help!