lgsvl/simulator #988: Socket Communication in python Api's

asifsid-32, 2020-09-16:
Hello LGSVL team,

I am trying to pass the camera output images from the simulator python api to my framework.

Can I use socket communication between python api and my framework, python api being the server ?

ntutangyun, 2020-09-16:
Hi,

If your framework is written in Python, you could install the Python API package and use the API to get the camera sensor outputs.

Check out the [quick start example](https://github.com/lgsvl/PythonAPI/blob/master/quickstart/06-save-camera-image.py).

In this scenario, the LGSVL simulator is the WebSocket server, and your framework with Python API installed is the client.

If your framework is not written in Python, the language you use should also provide a way to create a WebSocket client and connect to LGSVL simulator. You could check out the [Python API source code](https://github.com/lgsvl/PythonAPI) for more information on how to format commands and parse the response. 

---

Or if you prefer the way LGSVL communicate with Apollo, where Apollo (cyber bridge) is a socket server and LGSVL is a client,
You could also set up a socket server (i.e. mimic cyber bridge) in your framework, and let LGSVL write camera sensor images to your framework via this socket connection. 

Check out [here](https://github.com/ApolloAuto/apollo/tree/master/modules/contrib/cyber_bridge), especially `client.cc` ([link](https://github.com/ApolloAuto/apollo/blob/master/modules/contrib/cyber_bridge/client.cc)), on how to parse the binary data from LGSVL simulator.

asifsid-32, 2020-09-16:
> Hi,
>
> If your framework is written in Python, you could install the Python API package and use the API to get the camera sensor outputs.
>
> Check out the [quick start [...]

Can i make one of python api as server for example 06-save-camera-image.py which will start simulator as well as a connection with my framework i.e client ? I tried doing that but facing some errors !! I added my driver server code to 06-save-camera-image.py and tried running it but facing errors .. 

Following errors i got - > 

Traceback (most recent call last):
  File "102-save_camera_image.py", line 10, in <module>
    import lgsvl
  File "/Downloads/cpp-python_socket-image_transferring/lgsvl/__init__.py", line 8, in <module>
    from .simulator import Simulator, RaycastHit, WeatherState
  File "/Downloads/cpp-python_socket-image_transferring/lgsvl/simulator.py", line 7, in <module>
    from .remote import Remote
  File "/Downloads/cpp-python_socket-image_transferring/lgsvl/remote.py", line 35
    async def process(self):
            ^
SyntaxError: invalid syntax

asifsid-32, 2020-09-18:
> Hi,
>
> If your framework is written in Python, you could install the Python API package and use the API to get the camera sensor outputs.
>
> Check out the [quick start [...]

Hii @ntutangyun and @EricBoiseLGSVL I was able to send single image using socket communication to my framework. If I want to send every frame directly without first writing it to the storage to my framework which variable should I use which will have all the image frame stored in it ??

ntutangyun, 2020-09-18:
> first writing it to the storage to my framework

What do you mean by this, what is your storage.

> have all the image frames stored in it ??

the images are generated per frame and should also be transferred one by one. 
If you intend to a history of received images, you should save the images in a list or array

asifsid-32, 2020-09-18:
> What do you mean by this, what is your storage.
>
>
> the images are generated per frame and should also be transferred one by one.
> If you intend to a history of received images, you should save the [...]

We are interested in getting frame buffer handle. 
We dont want to use  " s.save("main-camera.png", compression=0) " instead we want an api which should be able to give us pointer to the frame buffer.

martins-mozeiko, 2020-09-18:
That is not possible due to the way how modern GPU APIs's work. You cannot get pointer to framebuffer. All you can do is copy data from framebuffer to CPU memory, which then you can access on CPU. That is pretty much what `save` method on `CameraSensor` is doing - additionally it is also compressing to jpeg or png for more efficiency (to transfer fewer bytes to Python). If you don't want compression, you can modify CameraSensor.cs source code to do whatever you went with pixel data that was copied from GPU.

reinaldymaslim, 2021-11-02:
Hello, I'm using lgsvl for my project,

I would like to retrieve the frame by frame front camera image. Does ego.get_sensors() returns the sensor properties for every frame in the simulation? my current code is the following. Thank you

counter=1
for sensor in ego.get_sensors():

    if sensor.name == "Main Camera":
        sensor.save("img/img"+str(counter)+".jpg",compression=8)
        counter+=1

EricBoiseLGSVL, 2021-11-02:
@reinaldymaslim 
API commands are called in update and will be executed per frame.  We recommend using the bridge for data collection.

reinaldymaslim, 2021-11-03:
Hi @EricBoiseLGSVL thank you for the reply. Could you please elaborate on this ros bridge? I sill don't know how to start using bridge. I have installed ros1 melodic. I'm using PythonAPI and the agent is Lincoln2017mkz with Apollo 5.0. For now I'm using step-pause method to obtain my front camera images which I set to save in a folder.

EricBoiseLGSVL, 2021-11-03:
Sensors can have publishers that you can subscribe to for data.  See [here](https://www.svlsimulator.com/docs/tutorials/ground-truth-obstacles/) for ground truth example.