lgsvl/simulator #993: Issue in integrating CAN-Bus and Utrasonic sensor on ROS

User-712, 2020-09-20:
Hello LGSVL team,
I have a question about using CAN-Bus and Ultrasonic sensor in LGSVL on ROS.
[image]
[image]
It looks like Canbus Data and Utrasonic datatypes are not supported in ROS format for ROS brdge. Could you advise how to publish canbus and Ultrasonic sensor data through rosbridge, so that i can subscribe it in ROS. 
How can we include the canbus and ultrasonic messages in ROS?
I don't wan't to use autoware or apollo.

Thanks!

rongguodong, 2020-09-21:
The current Ultrasonic sensor is only used in ROS2 bridge. If you want to use it with ROS or Cyber bridge, you need: 1) register the corresponding publisher in the corresponding bridge factory; and 2) create the convertor to convert UltrasonicData to corresponding data structure.

You can refer to [this commit](https://github.com/lgsvl/simulator/commit/35b476526ab7b755b2c28b493a9c1bee02f5a858) to see how to did this for ROS2 bridge. We will consider to add ROS/Cyber support in future release.

tajinder96, 2020-10-05:
Hi! I noticed that you only answered the question for the ultrasonic sensor. Is it the same for the can-bus? Also, I am guessing the steps needed to use the can-bus sensor with ROS look similar to the ones you mentioned above for the ultrasonic sensor?

closed 2020-10-20 by EricBoiseLGSVL