lgsvl/simulator #994: can not establish api port

qwetqwe, 2020-09-21:
Hi，team，
I want to use 2  PC to run simulation. One is for apollo stack,another is for simulator.
My apollo PC IP is 192.168.1.102
simulator PC is 192.168.1.99
 I change the config.yml to
api_hostname：“192.168.1.99”
api_port:8181
Sometimes it can establish port,sometimes not. I'm confused. I use 2020.06 release version
my ufw is inactive and I can use nc -l 8181 to establish listening port well.

EricBoiseLGSVL, 2020-09-21:
@shalinmehtalgsvl Do you have any ideas why this is happening?

lemketron, 2020-09-21:
Are you running Linux on both or just Apollo? Ok which machine are you running python scripts? Have you tried running python scripts on the simulator machine? If simulator machine is Windows there may be a firewall enabled that is blocking external access from the script running on the Apollo machine.

EricBoiseLGSVL, 2020-09-21:
@qwetqwe can you list all ports being listened to?  Maybe you have a conflict?

qwetqwe, 2020-09-29:
maybe network problem

closed 2020-09-29 by qwetqwe