lgsvl/simulator #995: the NPC can't be static

qwetqwe, 2020-09-21:
Hi team,
when I use api to control simulator, I add a NPC and api code is like this. But after a while, the NPC run by it self even if I set velocity to zero and follow_closest_lane to false.
```
import os
import lgsvl
import math
sim = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "localhost"), 8181)
if sim.current_scene == "BorregasAve":#SanFrancisco
  sim.reset()
else:
  sim.load("BorregasAve")
orientation=lgsvl.Vector(0,0,0)
tr=sim.map_from_gps(northing=4141514,easting=587033.53,altitude=0,orientation=0)
print(tr)
spawns = sim.get_spawn()
state = lgsvl.AgentState()
state.transform = tr
state.transform.rotation=spawns[0].rotation
forward = lgsvl.utils.transform_to_forward(spawns[0])
print("foward state:",forward)
state.velocity = 0 * forward
#state.transform=spawns[0]
a = sim.add_agent("vehicle_Lincoln2017MKZ", lgsvl.AgentType.EGO, state)
a.connect_bridge("192.168.1.102", 9090)
state.rotation.y=state.rotation.y
state.velocity = 0 * forward
state.transform.position=state.transform.position+30*forward
npc1 = sim.add_agent("Sedan", lgsvl.AgentType.NPC, state)
npc1.follow_closest_lane(False, 0)
sim.run(time_limit = 0)
```

ntutangyun, 2020-09-21:
what happens if you remove this lane `npc1.follow_closest_lane(False, 0)`

qwetqwe, 2020-09-21:
@ntutangyun  it is the same

EricBoiseLGSVL, 2020-09-21:
@qwetqwe We will look at this to see what is going on, thanks

hadiTab, 2020-09-21:
@qwetqwe I was able to reproduce this with your script. We will look into it. Thanks.

shalinmehtalgsvl, 2020-09-21:
Hi @qwetqwe ,

I believe this is just because the NPC is getting a non-zero speed after being dropped. I edited your script to use the raycast function and get points on the ground where the NPC can be spawned.

Can you try this change? This block replaces the line `state.transform.position=state.transform.position+30*forward`
```
# state.transform.position=state.transform.position+30*forward
up = lgsvl.utils.transform_to_up(tr)
layer_mask = 0
for bit in [0, 10, 11, 12]: # do not put 9 here, to not hit EGO vehicle itself
  layer_mask |= 1 << bit
hit = sim.raycast(state.transform.position+30*forward, -up, layer_mask)
state.transform.position = hit.point
```

qwetqwe, 2020-09-22:
@shalinmehtalgsvl ,
 it works,and car doesn't move. I thought that `npc1.follow_closest_lane(False, 0)` can brake the car.
thanks a lot

closed 2020-09-22 by qwetqwe