lgsvl/simulator #997: Problem with creating new NPC model

gyp199009, 2020-09-21:
Hello,
I am now creating a new NPC model under instruction by [https://www.lgsvlsimulator.com/docs/npc-plugins/#creating-npc-model-plugins] , and inside my NPC prefab I have Lights with names: LightHeadRight, LightHeadLeft, LightBrakeLeft, LightBrakeRight, IndicatorReverse, IndicatorLeftRear, IndicatorRightRear, IndicatorLeftFront, IndicatorRightFront, I also have a 3D model with parts: WheelLeftFront, WheelLeftRear, WheelRightFront, WheelRightRear, Body. But when I use the NPC plugin built from this prefab with a python script, it bounces up and down after the simulation runs, and what makes things more weird is after I substituted the part "Body" with a Cube created inside Unity editor (renamed to "Body"), everything is fine, while I substituted with a Cube (also named "Body") created  by Blender,  it went crazy again, so, what's the difference between models made by Unity editor and Blender?

EricBoiseLGSVL, 2020-09-21:
@gyp199009 You most likely have the pivots of the wheels in the wrong place.  Be sure to have the wheels pivot centered in the middle of the wheel.  Also, be sure to zero the other mesh pivots at origin.  Once unity runs, it creates the wheel colliders and attaches them to the wheel mesh.  If these are off then it can cause all kinds of issues.
Unity meshes and Blender are pretty much the same except for:
1) Export from blender has different orientation.  Unity wants Z forward.  Look up orientation differences in modeling/ROS/Unity
2) Unity primitives have a center pivot, not origin.  You will notice that the root NPC object is centered at origin.  You can fix this by having the primitive as a child to place in the right spot but it can cause issues
3) You can only change the scale of a Unity primitive to change it's size, this will cause issues with physics.  You want a actor to be at scale 1:1.  This also could be causing an issue.
Hope this helps and please send any images of the pivots in Blender and Unity to help find the issue.

gyp199009, 2020-09-22:
@EricBoiseLGSVL  I don't think it's about the wheels, but the "Body", because the same wheels with different bodies get different results, and I think the pivots of bodies are both at the geometric center.
Here is the body created in Unity:
[image]
And here is the body created in Blender:
[image]
I really can't see any difference between them, but results show one is fine and the other is absolutely abnormal!

EricBoiseLGSVL, 2020-09-22:
Can you please select the wheels so we can see the pivots?  It should look like this and at this angle to see relative to other objects
[image]

I think I see the problems.
- First, make sure there is no collider on the body.  I can't tell bc the image cuts off the Inspector view.  If there is, please remove it.
- Second, I added a cube like you at 0,0,0.  If you rotate the view, you can see that a cube (unity primitive always has center pivot) and it is below the wheel positions.  You will need to move it up

[image]

gyp199009, 2020-09-23:
@EricBoiseLGSVL  I found the reason, the  "Read/Write Enabled" on the inspector panel of model Import Settings should be checked!
[image]

EricBoiseLGSVL, 2020-09-23:
@gyp199009 Yes sorry, we toggled that on so long ago for npc meshes.  I'll make a note to add to the documentation.  Thanks for finding this.