lgsvl/simulator #1001: added vehicle can't move with keyboard

chongxing, 2020-09-22:
I am trying to add a new vehicle with the following two tutorials. 
https://github.com/lgsvl/simulator/issues/401 
https://www.youtube.com/watch?v=R_Z65kimCII

what i am doing is to learn copy a Jaguar Car. I can successfully copy the vehicle ( I name it Car2).  But it doesn't react to keyboard-up-left-right event, and don't move.

I am working on Ubuntu, the Unity version is 2019.4.10f1. 

https://www.youtube.com/watch?v=R_Z65kimCII, by following this link, one thing I can't do, is to add BaseLink component. There is no such link inside the simulator.

[image]

by comparing, it works fine if use the original Jaguar Carhttps://assets.lgsvlsimulator.com/d49c88a93b11b047a3e21e0c7d4b2ec6964fb16d/vehicle_Jaguar2015XE)

EricBoiseLGSVL, 2020-09-23:
@chongxing 
- First off latest simulator is 2019.3.15f1.  Please use this version for 06 release.
- Be sure you have pulled the branch 06 release, not master.  BaseLink is a class in the project you add to an empty gameobject.  Place it where you need on the vehicle and add the BaseLink.cs component.  Then drag the transform to the BaseLink public reference on the VehicleSMI class in the inspector. Save and build vehicle asset bundle.
- Be sure to toggle off the mesh renderer for the collider mesh, it is only for the MeshCollider component (toggle convex on)
- To be able to drive manually, you need to add a KeyboardController sensor in the vehicle sensor config.

chongxing, 2020-09-23:
@EricBoiseLGSVL  Thanks for your quick reply. I will learn and have a try later.