lgsvl/simulator #1002: Issue in loading map for autoware.ai with lgsvl

User-712, 2020-09-22:
Hi,

I am running latest LGSVL simulator and autoware 1.12.0, facing issue in loading map. I have followed the instructions given here https://www.lgsvlsimulator.com/docs/autoware-instructions/
https://github.com/Autoware-AI/autoware.ai/wiki/Generic-x86-Docker#run-an-autoware-docker-container

1. ./run.sh -t 1.12.0 in docker/generic directory
2. sudo apt update && sudo apt install ros-$ROS_DISTRO-image-transport-plugins -y
3. roslaunch runtime_manager runtime_manager.launch in Autoware directory in container. 
[image]
4. As soon as I upload my_map.launch file in quick start, dialog box provides an error as in the image below.
[image]
And not getting any map in Rviz
3. If I select fixed frame as velodyne, am getting below output in Rviz
[image]
[image]

Creating a Simulation choosing BorregasAve map and Jaguar2015XE (Autoware). Running the created Simulation.

I also tried with 93d92d6becc8dbced0a588aadcf0cee18e2c2429 commit, for this I was able to get map but not able to get pointcloud, vehicle and other data on the map.

System specification:
OS:Linux 16.04 (memory: 31.3 GiB, processor: core i7, Graphics: GeForce GTX)
Simulator: LGSVL 2020.01 (released version)

EricBoiseLGSVL, 2020-09-23:
@User-712 Autoware 1.12 is not supported.  Please use 1.14.  Right @hadiTab ?

hadiTab, 2020-09-23:
1.12 is partially supported, but we encourage using 1.14 instead. You should be able to load the maps in 1.12 though. @cyberphysicalrobotics do you know what this is about? Thanks.

jiameixie, 2020-09-29:
@hadiTab I used 1.14 and followed the steps in https://www.lgsvlsimulator.com/docs/autoware-instructions/.When I load map and detect, below errors occurred.   Need I set the default.rviz? 

[ERROR] [1601360598.734482481]: [createSignalMarkerArray] failed createPoleMarker: 4,4,6.75,0.4
RLException: while processing 

/home/autoware/Autoware/install/vision_ssd_detect/share/vision_ssd_detect/launch/vision_ssd_detect.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/autoware/Autoware/install/vision_ssd_detect/share/vision_ssd_detect/launch/vision_ssd_detect.launch'
The traceback for the exception was written to the log file

hadiTab, 2020-09-29:
@jiameixie both of those errors can be safely ignored.

Have you gone through the rest of the steps and had any issues?

jiameixie, 2020-09-29:
@hadiTab Yes, I have one. I run lgsvl simulator on my window machine and autoware.ai on a different ubuntu machine. It seems the lgsvl simulator can't connect the autoware bridge port 9090.  I set the Vehicles IP to autowarehostIP:9090, where autowarehostIP is the IP address where running autoware container.  The ref https://www.lgsvlsimulator.com/docs/autoware-instructions run both simulator and autoware on the same machine.  Need I set the autoware container to export port 9090?

jiameixie, 2020-09-29:
@hadiTab The screenshot of my rivz.  It doesn't have a car on it, which is different from the demo.
[image]

jiameixie, 2020-09-29:
It's no need extra setup for the port. The container use the host network.
[image]

hadiTab, 2020-10-20:
@jiameixie can you show the list of topics being published? You can get that with `rostopic list`. Also, in rviz, there is a warning on TF and an error on Vehicle model. What do those say?

jiameixie, 2020-10-21:
@hadiTab It's unstable to run lgsvl with autoware. It was a different situation every time I run it. So I can't get the list of topics now which lead the above errors.   
I got help from someone. He told me that some parameters in launch files are wrong and need manual setting. The autoware version he used is 1.13.0. My autoware version is 1.14.0. Following the steps he provided , the vehicle in lgsvl can move along with the vehicle in autoware.  But sometime it went wrong. The vehicle just moved forward straight without stop or twist. Sometimes it can run normally. It made me confused.
what we did is to specify its pose location by command lines.
-  Add  {"type": "Clock", "name": "Clock Sensor","params": {"Topic": "/clock"} } to lgsvl vehicle setup. 
- Reload map and run `rosparam set /use_sim_time false`.
- For Setup tab in runtime manager,  click enable Vehicle Model
- For Computing tab, click waypoint_planner/astar_avoid, waypoint_planner/velocity_set, waypoint_follower/pure_pursuit and waypoint_follower/twist_filter. 
-  Set vehicle pose. `rostopic pub /initialpose geometry_msgs/PoseWithCovarianceStamped --once -- '{ header: {stamp: now, frame_id: "world"}, pose:{pose:{position: {x: -7.44751739502, y: 51.3522491455, z: 0.0},orientation: {x: 0.0, y: 0.0, z: -0.789084084512, w: 0.60506553654}}}}'`

Do you think the above steps necessary to run lgsvl with autoware.ai? @hadiTab

hadiTab, 2020-10-23:
@jiameixie the above steps are similar to what is in the default launch files. Except for adding the clock sensor, which helps avoid delays when the performance is not good. Manually publishing the pose, should be the same as just marking the initial pose in rviz as well. The only reason I can think of that this is working for you but the launch files aren't would be that one of those nodes is crashing when the launch files is used, but doesn't crash when enabled manually. I've seen that happen before with some nodes in Autoware.