lgsvl/simulator #1004: Data frame conversion from IMU to ODOM

Mojojojooo, 2020-09-23:
1.How are the frame data of IMU and ODOM related. 
2.What can be done in the source code to convert data from IMU frame to ODOM frame.

hadiTab, 2020-09-23:
@Mojojojooo I'd suggest reading [this page](https://www.lgsvlsimulator.com/docs/sensor-json-options/) from our documentation which explains the how to configure the sensors on the vehicle. The relative location of the sensors is defined by the `transform` in the JSON config. By default, if no `parent` is defined for the sensor, the transform will be relative to the center of the vehicle projected onto the ground. However you can add a `parent` entry for each sensor which will define where the transform is relative to. If you would like to have a random point defined in the vehicle to use as a parent for other frames, you can use the [Transform Sensor](https://www.lgsvlsimulator.com/docs/sensor-json-options/#transform-sensor) to create such a reference point and use it as `parent` for other sensors.

Mojojojooo, 2020-09-24:
So if we define the transform sensor at the center of the car and add parent entery to all other sensors then everything will be with respect to the center. Did I get it correct ?

hadiTab, 2020-09-24:
Yes, that is correct.

closed 2020-09-27 by Mojojojooo