lgsvl/simulator #1005: How is the ODOM frame defined

Mojojojooo, 2020-09-23:
With respect to what is the odom frame made. (vehicle center or the rear axle of the car )

hadiTab, 2020-09-23:
@Mojojojooo frames are defined by the user in the JSON configuration file in the web-ui. Can you share the JSON configuration you are using?

Mojojojooo, 2020-09-24:
{
	"name": "simulator-web-app",
	"version": "1.0.0",
	"description": "A WebUI application for LGSVL Simulator.",
	"author": "",
	"main": "src/index.js",
	"scripts": {
		"serve": "enact serve -p 8081",
		"pack": "enact pack",
		"pack-p": "enact pack -p",
		"watch": "enact pack --watch",
		"clean": "enact clean",
		"lint": "enact lint .",
		"license": "enact license",
		"test": "enact test start --single-run",
		"test-watch": "enact test start"
	},
	"license": "UNLICENSED",
	"private": true,
	"repository": "",
	"enact": {
		"theme": "moonstone",
		"proxy": "http://localhost:8080",
		"ri": {
			"baseSize": 16
		}
	},
	"eslintConfig": {
		"extends": "enact"
	},
	"eslintIgnore": [
		"node_modules/*",
		"build/*",
		"dist/*"
	],
	"dependencies": {
		"@enact/cli": "^2.8.1",
		"@enact/core": "^2.5.3",
		"@enact/i18n": "^2.5.3",
		"@enact/moonstone": "^2.5.3",
		"@enact/spotlight": "^2.5.3",
		"@enact/ui": "^2.5.3",
		"axios": "^0.19.2",
		"classnames": "^2.2.6",
		"es6-promise": "^4.2.8",
		"eventsource": "^1.0.7",
		"eventsource-polyfill": "^0.9.6",
		"prop-types": "^15.6.2",
		"react": "^16.13.1",
		"react-datepicker": "^2.16.0",
		"react-dom": "^16.13.1",
		"react-favicon": "0.0.16",
		"react-icons": "^3.10.0",
		"react-router-dom": "^5.2.0",
		"url-parse": "^1.4.7"
	}
@hadiTab This is the content of the package.json file.

hadiTab, 2020-09-24:
This is not the correct JSON. Once you start the simulator, click on the button to open the web ui. Inside the webui click on the vehicles tab. Each vehicle has a wrench icon. Click on the icon and a JSON formatted sensor configuration will be brought up for that vehicle.

Mojojojooo, 2020-09-24:
[{"type": "GPS Device", "name": "GPS",
            "params": {"Frequency": 12.5, "Topic": "/nmea_sentence", "Frame": "gps", "IgnoreMapOrigin": true},
            "transform": {"x": 0, "y": 0, "z": 0, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "GPS Odometry", "name": "GPS Odometry",
            "params": {"Frequency": 12.5, "Topic": "/odom", "Frame": "gps", "IgnoreMapOrigin": true},
            "transform": {"x": 0, "y": 0, "z": 0, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "IMU", "name": "IMU",
            "params": {"Topic": "/imu_raw", "Frame": "imu"},
            "transform": {"x": 0, "y": 0, "z": 0, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Lidar", "name": "Lidar",
            "params": {"LaserCount": 32, "MinDistance": 0.5, "MaxDistance": 100, "RotationFrequency": 10, "MeasurementsPerRotation": 360, "FieldOfView": 41.33, "CenterAngle": 10, "Compensated": true, "PointColor": "#ff000000", "Topic": "/points_raw", "Frame": "velodyne"},
            "transform": {"x": 0, "y": 2.312, "z": -0.3679201, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Color Camera", "name": "Main Camera",
            "params": {"Width": 1920, "Height": 1080, "Frequency": 15, "JpegQuality": 75, "FieldOfView": 50, "MinDistance": 0.1, "MaxDistance": 1000, "Topic": "/simulator/camera_node/image/compressed", "Frame": "camera"},
            "transform": {"x": 0, "y": 1.7, "z": -0.2, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Keyboard Control", "name": "Keyboard Car Control"},{"type": "Wheel Control", "name": "Wheel Car Control"},{"type": "Vehicle Control", "name": "Autoware Car Control",
            "params": {"Topic": "/vehicle_cmd"} }] 

Is this the one ?

hadiTab, 2020-09-24:
Yup. That's the one.

closed 2020-09-27 by Mojojojooo