lgsvl/simulator #1009: Fisheye correction Model

asifsid-32, 2020-09-24:
Hii LGSVL team,

I am working with fisheye camera. For calibrating it I need to know whether the fisheye correction Model used is kannala ?

If not what is the correction Model used ?

rongguodong, 2020-09-24:
It is based on "Unified Projection Model" (http://www.robots.ox.ac.uk/~cmei/articles/single_viewpoint_calib_mei_07.pdf)

asifsid-32, 2020-09-24:
> Hii LGSVL team,
> 
> I am working with fisheye camera. For calibrating it I need to know whether the fisheye correction Model used is kannala ?
> 
> If not what is the correction Model used ?

   Can you also tell me the fish eye camera model used for LGSVL simulator ??

rongguodong, 2020-09-24:
Please refer the above paper for more details, or simply Google "Unified Projection Model".

asifsid-32, 2020-09-25:
> Please refer the above paper for more details, or simply Google "Unified Projection Model".

Can you please tell me Focal length in pixels of fisheye camera and sensor size of the fisheye camera ?

rongguodong, 2020-09-25:
> > Please refer the above paper for more details, or simply Google "Unified Projection Model".
> 
> Can you please tell me Focal length in pixels of fisheye camera and sensor size of the fisheye camera ?

This depends on the real camera you are trying to simulate. For a real camera, if you use unified projection model to calibrate, the calibration app should return all the values you need for this sensor. Please refer to the paper for more details.

If you are talking about our example JSON, the image size is 1920x1080, and the focal length can be calculated from image size and FOV (i.e. f = Height / 2 / tan (FieldOfView / 2)).

asifsid-32, 2020-09-25:
> Please refer the above paper for more details, or simply Google "Unified Projection Model".

@rongguodong do you have some api or mechanism for fish correction/ getting fisheye corrected images ??

rongguodong, 2020-09-25:
> > Please refer the above paper for more details, or simply Google "Unified Projection Model".
> 
> @rongguodong do you have some api or mechanism for fish correction/ getting fisheye corrected images ??

I used some opensource code which can do this. But I do not have it right now. If I can find it, I will post links here. You can try to Google online in the meanwhile.

rongguodong, 2020-09-25:
https://github.com/hengli/camodocal

asifsid-32, 2020-09-25:
> > > Please refer the above paper for more details, or simply Google "Unified Projection Model".
> > 
> > Can you please tell me Focal length in pixels of fisheye camera and sensor size of the fisheye camera ?
> 
> This depends on the real camera you are trying to simulate. For a real camera, if you use unified projection model to calibrate, the calibration app should return all the values you need for this sensor. Please refer to the paper for more details.
> 
> If you are talking about our example JSON, the image size is 1920x1080, and the focal length can be calculated from image size and FOV (i.e. f = Height / 2 / tan (FieldOfView / 2), which is around 1158 for our example JSON).

@rongguodong This formula is for pinhole model ....

rongguodong, 2020-09-25:
> @rongguodong This formula is for pinhole model ....

Yes, the unified projection model is in fact a pinhole model with shifted projection center (by xi). Please refer to the paper for more details. The opensouce code CamOdoCal may also help.

asifsid-32, 2020-09-28:
> Yes, the unified projection model is in fact a pinhole model with shifted projection center (by xi). Please refer to the paper for more details. The opensouce code CamOdoCal may also help.

@rongguodong Using the DistortionParameters provided in json sensor configs how can we convert the fish_eye points into the pinhole camera models ? What specific algorithm should i refer in the paper ?

closed 2020-12-02 by EricBoiseLGSVL