lgsvl/simulator #1010: Need Help in Modular Testing

SungYinYang, 2020-09-24:
My GPU is RTX2080S.
I was running Apollo master pre6 branch

I follow the video from https://www.youtube.com/watch?time_continue=6&v=781zkzN2xMg&feature=emb_logo to do the modular testing.

When I enable localization on the modular tab, it has the following error.

Failed to load default POI. Please make sure the file exists at /apollo/modules/map/data/borregas_ave/default_end_way_point.txt

[image]

Furthermore, I was able to open my dreamview even the terminal tell me failed to start Dreamview. Is the above problem related to dreamview? 

[image]

Any help and recommendation is appreciate!
Thanks in advance.

rongguodong, 2020-09-24:
For the error of POI, it seems having no problem to leave with it.
For dreamview error, as instructed here, please check the log file to find out the exact reason.

EricBoiseLGSVL, 2020-09-25:
@rongguodong can you edit your comment to include the link? thanks

rongguodong, 2020-09-25:
> @rongguodong can you edit your comment to include the link? thanks

When I said "as instructed here", I meant the instruction in the screenshot of the console output, which said "Please check /apollo/data/log". :)

SungYinYang, 2020-09-25:
@rongguodong Thanks for answering my question fast!
I see the following error in my dreamview.info log.

      disk_spaces {
        path: "/apollo/data"
        insufficient_space_warning: 8
        insufficient_space_: 2
      }
      disk_spaces {
        path: "/media/apollo/internal_nvme"
        insufficient_space_warning: 128
        insufficient_space_error: 32
      }
    }

Any suggestion on what file that I should delete?

rongguodong, 2020-09-25:
So, you do not have enough space on hard drive?

SungYinYang, 2020-09-25:
My hardrive have enough space. It has free space more than 178.1 GB.
Do recommend me to delete the files in /apollo/data/log and /apollo/data/core ?

rongguodong, 2020-09-25:
If you have enough space and dreamview report insufficient space, it sounds like a Dreamview issue. Please report it to Apollo.

SungYinYang, 2020-09-25:
@rongguodong 
I am able to open my dreamview and link it to LGSVL.
I am also able to set an route and run the apollo car with perception and traffic light on.
Thus, I think I would ignore dreaview failuer in my terminal.

I follow the video again and find out that my dreamview was not able to detect traffic light signal as video show.

[image]

Any suggestion that should double check on?
Thanks in advance!

Below is the json that I put in the cyberRT configuration.

[{
    "type": "3D Ground Truth",
    "name": "3D Ground Truth",
    "params": {
      "Frequency": 10,
      "Topic": "/apollo/perception/obstacles"
    },
    "transform": {
      "x": 0,
      "y": 1.975314,
      "z": -0.3679201,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
},{
    "type": "Signal",
    "name": "Signal Sensor",
    "params": {
      "Frequency": 10,
      "MaxDistance": 100,
      "Topic": "/apollo/perception/traffic_light"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": 0,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
},

  {
    "type": "CAN-Bus",
    "name": "CAN Bus",
    "params": {
      "Frequency": 10,
      "Topic": "/apollo/canbus/chassis"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": 0,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "GPS Device",
    "name": "GPS",
    "params": {
      "Frequency": 12.5,
      "Topic": "/apollo/sensor/gnss/best_pose",
      "Frame": "gps"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": -1.348649,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "GPS Odometry",
    "name": "GPS Odometry",
    "params": {
      "Frequency": 12.5,
      "Topic": "/apollo/sensor/gnss/odometry",
      "Frame": "gps"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": -1.348649,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "GPS-INS Status",
    "name": "GPS INS Status",
    "params": {
      "Frequency": 12.5,
      "Topic": "/apollo/sensor/gnss/ins_stat",
      "Frame": "gps"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": -1.348649,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "IMU",
    "name": "IMU",
    "params": {
      "Topic": "/apollo/sensor/gnss/imu",
      "Frame": "imu",
      "CorrectedTopic": "/apollo/sensor/gnss/corrected_imu",
      "CorrectedFrame": "imu"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": -1.348649,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "Radar",
    "name": "Radar",
    "params": {
      "Frequency": 13.4,
      "Topic": "/apollo/sensor/conti_radar"
    },
    "transform": {
      "x": 0,
      "y": 0.689,
      "z": 2.272,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "Keyboard Control",
    "name": "Keyboard Car Control"
  },
  {
    "type": "Wheel Control",
    "name": "Wheel Car Control"
  },
  {
    "type": "Vehicle Control",
    "name": "Apollo Car Control",
    "params": {
      "Topic": "/apollo/control"
    }
  }
]

rongguodong, 2020-09-25:
Are you talking about "NO SIGNAL" on top right corner? It seems not indicating if Apollo received signal result or not.
As long as you can see the traffic light changing colors in Dreamview, it means Apollo HAS received signal result.
@daviduhm Do you have more ideas on that "NO SIGNAL"? I think I have seen that sometime as well.

SungYinYang, 2020-09-25:
Yes, I am talking about the No signal on the top right corner.
I am not able to see the traffic light changing colors in Dreamview. 
My car would crash into the traffic.

Would you mind posting your setting of JSON file in the comment?
Thanks in advance.

I would give you more feedback when I find some possible errors.

rongguodong, 2020-09-25:
Your JSON seems fine. Particularly, I see you have "Signal Sensor". Please confirm that `/apollo/perception/traffic_light` is green in cyber_monitor. If so, you should see the traffic lights changing color in Dreamview. For example, they are all red in your screenshot (i.e. the red dots). And they should become green when they become green in the simulator.

EricBoiseLGSVL, 2020-09-25:
Don't you need a telephoto sensor for light detection?

rongguodong, 2020-09-25:
> Don't you need a telephoto sensor for light detection?

He is using modular testing, where the traffic light result is generated by simulator, not the perception module. If you want perception module to detect the traffic light, you do need a tele camera.

lemketron, 2020-09-25:
> I am also able to set an route and run the apollo car with perception and traffic light on.

Is the "traffic light" module enabled? If so, you should turn it off since it seems you are using the "signal sensor" in the simulator which publishes the signal information on the `/apollo/perception/traffic_light` topic. 

As noted above, check cyber_monitor; specifically you should locate and expand the `/apollo/perception/traffic_light` topic as shown below, where you can confirm that the traffic light color information is being sent by the simulator to Apollo:

[image]

daviduhm, 2020-09-28:
@SungYinYang With an assumption that you have set up modular testing correctly following [this](https://www.lgsvlsimulator.com/docs/modular-testing/) documentation, "NO SIGNAL" message means that either you're not running Apollo's planning module or your planning module is not working properly. On the screenshot you posted above, I don't see a green curvy band coming from the ego vehicle in Dreamview like below meaning that your planning is not running. Without running the planning module, Dreamview will show the "NO SIGNAL" message on the top right.

[image]