lgsvl/simulator #1011: [documentation] Traffic behavior plugin drunkDriver link is 404

ntutangyun, 2020-09-24:
Hi team,

at this [documentation page](https://www.lgsvlsimulator.com/docs/npc-plugins/)

just found that the drunk driver link [(https://github.com/lgsvl/ExampleNPCs/DrunkDriver)](https://github.com/lgsvl/ExampleNPCs/DrunkDriver) page cannot be found.

Also there seems no such repositories for ExampleNPCs.

ntutangyun, 2020-09-24:
Hi team, 

following the instructions [here](https://www.lgsvlsimulator.com/docs/npc-plugins/#npc-behaviors)

I've created a custom script named `ManualLaneFollow.cs` and placed in below folder `Assets/External/NPCs/CustomNPCControllers/ManualLaneFollow/ManualLaneFollow.cs`
with following content:

```c#
...

public class NPCManualLaneFollowBehavior : NPCLaneFollowBehaviour
{
    ...

    protected void GetNextLane()
    {
        ...
    }
}
```

And I created another api command `NPCManualLaneFollow.cs` and placed at 
`Assets/Scripts/Api/Commands/NPCManualLaneFollow.cs`
with following content:

```c#
...

namespace Simulator.Api.Commands
{

class NPCManualLaneFollow : ICommand
{
    public string Name => "npc/manual_lane_follow/config";

    public void Execute(JSONNode args)
    {
        var uid = args["uid"].Value;
        var api = ApiManager.Instance;

        if (!api.Agents.TryGetValue(uid, out GameObject npc))
        {
            api.SendError(this, $"Agent '{uid}' not found");
            return;
        }

        var behaviour = npc.GetComponent<NPCManualLaneFollowBehavior>();
        if (behaviour == null)
        {
            api.SendError(this, $"Agent '{uid}' is not a manual lane follow NPC agent");
            return;
        }
       ...
    }
}
}
```

The simulator produces this error:
The type or namespace name "NPCManualLaneFollowBehavior" count not be found...

EricBoiseLGSVL, 2020-09-25:
@ntutangyun Yes, sorry we missed releasing this DrunkDriver behavior in 06 release.  We will be releasing with the next one soon.
As for the code issue it looks fine. Did you create an asset bundle of the new behavior class you created?

closed 2020-09-26 by ntutangyun