lgsvl/simulator #1012: cannot launch standalone executable file on Ubuntu

InitialDuan, 2020-09-24:
Hello guys,

I downloaded lgsvlsimulator-linux64-2020.06 release from lgsvl website and run command line "sudo apt-get install libvulkan1". After I double-click the Simulator and then press Ok, nothing appears.

I have NVIDIA GeForce GTX 960M and driver 450.66 on Ubuntu18.04, and Nvidia-smi output looks also fine.
This is the content of /.config/unity3d/Editor/Player.log:

[Package Manager] Server::Start -- Port 46742 was selected

 COMMAND LINE ARGUMENTS:
/opt/Unity/Editor/Unity
-createproject
/home/my lgsvl
-cloneFromTemplate
/opt/Unity/Editor/Data/Resources/PackageManager/ProjectTemplates/com.unity.template.3d-2.3.3.tgz
-cloudOrganization
undefined
-cloudEnvironment
production
-useHub
-hubIPC
-hubSessionId
13293130-fdeb-11ea-8e90-8bf5b4999b05
-accessToken
wZVV2ssIkQOx57keRnBibgyHZa6GlzD45r-SxkzBuJ4008f
CreateDirectory '/home/my lgsvl' failed:  (current dir: /home/runtao/Downloads)
Creating project folder: /home/my lgsvl failed.

(Filename: ./Runtime/Utilities/Argv.cpp Line: 358)

Assertion failed on expression: '!projectPath.empty()'

(Filename: ./Editor/Src/ProjectUtility.cpp Line: 634)

LICENSE SYSTEM [2020924 0:37:50] Next license update check is after 2020-09-24T22:26:29

EricBoiseLGSVL, 2020-09-25:
@InitialDuan Our recommended spec for GPU is 8gb.  I think you are also not using the GPU.  It's hard to tell from your formatting of the NvidiaSMI please use ``` to format.  Please also look at multiple closed issues on Linux GPU setup.  I recommend a better GPU for running Simulator esp if you want to run AD stack on the same machine.

```
NvidiSMI
output
```

InitialDuan, 2020-09-25:
> @InitialDuan Our recommended spec for GPU is 8gb. I think you are also not using the GPU. It's hard to tell from your formatting of the NvidiaSMI please use ``` to format. Please also look at [...]

sorry, i updated the formatting of the NvidiaSMI. I am aware of the recommended spec for GPU, but 4gb only for simulator should at least work and not to lead to a immediate crash, right? Also, can Editor.log tell me something?

EricBoiseLGSVL, 2020-09-25:
Well, if you are running other applications and trying to run a large map, using integrated GPU instead of nvidia, then yes it will crash.  Still if the logs are empty then it seems that you are failing to actually run the simulator.  It looks like the you are failing to start the application.  Are you the only user on this machine?  I have seen other Linux issues with user issues and installing problems.  @luqiang21 any idea why the simulator is failing to install on Linux?

InitialDuan, 2020-09-25:
> Well, if you are running other applications and trying to run a large map, using integrated GPU instead of nvidia, then yes it will crash. Still if the logs are empty then it seems that you are [...]

Yes, i am the only user and runing only the simulator. The current situation is that after using the method mentioned in https://github.com/lgsvl/simulator/issues/753#issuecomment-626269278 to reinstall vulkan, i can now open the window and browser to configure the simulation, but when i try to click the red triangle button in browser to load the simluation, the window crashes again. And the log files look like the same as before.  

Also interesting is, when i switch to integrated GPU to test the simulator, it can load the image in the window and not crash, just can not start. So i believe the problem should rely on Nvidia GPU setup, but all the info like nvidia-smi, nvidia-setting, vulkaninfo and vkcube looks fine.

EricBoiseLGSVL, 2020-09-25:
Do you have mesa drivers installed?  if so, please remove and reinstall vulkan and see if that fixes it

lemketron, 2020-09-25:
> CreateDirectory '/home/my lgsvl' failed:  (current dir: /home/runtao/Downloads)
> Creating project folder: /home/my lgsvl failed.

Why is it trying to create a user directory called "my lgsvl"?

InitialDuan, 2020-09-25:
> > CreateDirectory '/home/my lgsvl' failed:  (current dir: /home/runtao/Downloads)
> > Creating project folder: /home/my lgsvl failed.
> 
> Why is it trying to create a user directory called "my lgsvl"?

em...It could come from when i try to add a new project on Unity Hub. But it should have no effect when i start the standalone executable file, right?

InitialDuan, 2020-09-25:
> Do you have mesa drivers installed? if so, please remove and reinstall vulkan and see if that fixes it

I did, but it only helps a little. As i mentioned above, reinstalling the vulkan allows me open the window and browser, but when i try to click the red triangle button in browser to load the simluation, the window crashes again.

EricBoiseLGSVL, 2020-10-06:
Have you been able to run without mesa drivers and find a solution for not being able to create the project folder?

lemketron, 2020-10-06:
Please (review and) share Player.log file after launching and crashing for possible insights...

InitialDuan, 2020-10-07:
> Please (review and) share Player.log file after launching and crashing for possible insights...

I checked it again after crashing, the Player.log is still empty. The only thing i observed is the following:
```
duan@ubuntu:~$ grep ^Vulkan ~/.config/unity3d/LG\ Silicon\ Valley\ Lab/LGSVL\ Simulator/Player.log
Vulkan detection: 2
Vulkan API version 1.1.0 (1.2.133 supported by driver)
Vulkan vendor=[NVIDIA] id=[10de]
Vulkan renderer=[GeForce GTX 960M] id=[139b]
Vulkan device type 2
Vulkan driverversion=[0x708cc180] uint=[1888272768]
Vulkan PSO: cache data not found [path=/tmp/LG Silicon Valley Lab/LGSVL Simulator/vulkan_pso_cache.bin]
```
Can "Vulkan PSO: cache data not found" tell something?

EricBoiseLGSVL, 2020-10-07:
What do you mean by "red triangle button in browser to load the simulation, the window crashes again".  Can you screen shot this please?  This still seems like you have issues on the OS even starting the application.  When you start the simulator you will get a playerlog.  Do you have the latest nvidia drivers?  Sorry, we haven't seen this issue before and any extra information would help to solve this.

InitialDuan, 2020-10-07:
> What do you mean by "red triangle button in browser to load the simulation, the window crashes again". Can you screen shot this please? This still seems like you have issues on the OS even starting [...]

[image]
Here is my screen shot. I did have a latest nvidia driver, as you can see from my nvidia-smi output.

EricBoiseLGSVL, 2020-10-07:
Yes, so you are running simulator and must have a player log.  Please find that and post, thanks

InitialDuan, 2020-10-07:
> Yes, so you are running simulator and must have a player log. Please find that and post, thanks
[image]

I take a screen shot again including viewing the Player.log. Can you check is that the right way?

InitialDuan, 2020-10-07:
> Yes, so you are running simulator and must have a player log. Please find that and post, thanks

sorry, i used to use gedit to view the player.log which looked always empty, now i just open the player.log in folder:
```
Mono path[0] = '/home/runtao/Downloads/lgsvlsimulator-linux64-2020.06/simulator_Data/Managed'
Mono config path = '/home/runtao/Downloads/lgsvlsimulator-linux64-2020.06/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Display 0 '0': 1920x1080 (primary device).
Loading player data from /home/runtao/Downloads/lgsvlsimulator-linux64-2020.06/simulator_Data/data.unity3d
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=17
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Vulkan detection: 2
Initialize engine version: 2019.3.15f1 (59ff3e03856d)
[Subsystems] Discovering subsystems at path /home/runtao/Downloads/lgsvlsimulator-linux64-2020.06/simulator_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=17
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] Graphics queue count=1
[Vulkan init] extensions: count=82
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_buffer_device_address, enabled=0
[Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_depth_stencil_resolve, enabled=0
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=1
[Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHR_draw_indirect_count, enabled=0
[Vulkan init] extensions: name=VK_KHR_driver_properties, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[... 1745 lines cut ...]
#19 0x0000000000000000 in ?? ()

=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

Caught fatal signal - signo:11 code:1 errno:0 addr:0x60
Obtained 14 stack frames.
#0  0x007f45c03998a0 in funlockfile
#1  0x007f45bb8ec113 in _nv043glcore
#2  0x007f45bb8fc167 in _nv043glcore
#3  0x007f45c127ea00 in vk::Image::CreateImageViews(VkDevice_T*, VkComponentMapping, VkFormat, bool, VkFormat)
#4  0x007f45c12b855e in vk::ImageManager::CreateImage(TextureDimension, int, VkExtent3D const&, GraphicsFormat, int, unsigned int, VkSampleCountFlagBits, vk::ImageMemory)
#5  0x007f45c1282ff1 in vk::Texture::Create(vk::CommandBuffer*, TextureID, TextureDimension, int, VkExtent3D const&, GraphicsFormat, int, bool, VkSampleCountFlagBits, vk::ImageMemory)
[... cut ...]
#7  0x007f45c12ae297 in vk::RenderSurface::CreateColorSurfacePlatform(vk::RenderSurface*, vk::CommandBuffer*, vk::ImageManager&, GraphicsFormat, bool, VkImage_T*)
#8  0x007f45c11f308c in GfxDeviceWorker::RunCommand(ThreadedStreamBuffer&)
#9  0x007f45c11fb72b in GfxDeviceWorker::RunExt(ThreadedStreamBuffer&)
#10 0x007f45c11f1305 in GfxDeviceWorker::RunGfxDeviceWorker(void*)
#11 0x007f45c15f5d3a in Thread::RunThreadWrapper(void*)
#12 0x007f45c038e6db in start_thread
#13 0x007f45c00b7a3f in clone
```

martins-mozeiko, 2020-10-08:
Crash on something like `vk::Image::CreateImageViews` most likely means that GPU is running out of memory. Unfortunately Unity likes to crash when it runs out of GPU memory instead of showing proper error message.

Try monitory GPU memory usage when BorregasAve is loading - run `watch -n 0.1 nvidia-smi -q -d MEMORY` in separate terminal. If you see free memory quickly approaching 0 - that's the problem.

Also try running with CubeTown or Single Lane Road maps - do they work. If yes, then your GPU doesn't have enough memory for BorregasAve.

InitialDuan, 2020-10-08:
> Crash on something like `vk::Image::CreateImageViews` most likely means that GPU is running out of memory. Unfortunately Unity likes to crash when it runs out of GPU memory instead of showing proper [...]

Thank you so much! You are right, free memory quickly approaching 0 leads to a crash. And running with Single Lane Road map does not lead to a crash, but awaiting in the loading forever... 
Obviously my GPU doesn't have enough memory for lgsvl, even though it runs smoothly on my win10 with same hardware. I opened a new issue for the help of connecting lgsvl on win10 and Autoware on virtual Machine https://github.com/lgsvl/simulator/issues/1044#issue-716635445.

closed 2020-10-08 by InitialDuan