lgsvl/simulator #1013: Wrong car position in Apollo 5.0

eyadshak, 2020-09-24:
So I am running Apollo and LGSVL on separate machines, I'm experiencing minimum latency when moving the car manually, and while observing the traffic light recognition, but the ego car's position in Apollo's Dreamview is wrong, causing the car to crash when control is enabled. It localizes in the wrong position but continues to mirror the movement of LGSVL with no delays, 

[image]
[image]

Here are the results of cyber_monitor

[image]

I checked more than one map and they all have the same issue, the HD map annotation tool shows annotations correctly in the unity editor and the NPCs are driving in the correct lanes.

Would anyone know how to fix this? I am running Apollo 5.0, for some reason when I tested it on the current Apollo master correct localization worked with no issue on Borregas Ave.

rongguodong, 2020-09-24:
I believe you are using 2020.06, which requires new map asset bundle. Please delete the database so that it will automatically download the new asset bundle. Or you can manually download it [here](https://content.lgsvlsimulator.com/maps/borregasave/).

eyadshak, 2020-09-27:
It was indeed a problem with using old maps, had to replace the asset bundle and HD map info (base_map, routing_map, sim_map).
Thank you for your help!

closed 2020-09-27 by eyadshak