lgsvl/simulator #1014: Depth Camera with floating point distance

mosshammer, 2020-09-24:
I'm trying to convert the fragment shader in DepthPass.hlsl to encode the distance inside the RGB(A) channels as a float similar to the lidar render pass. However, the fragment shader seems to be applied only to the NPCs and not the environment, it's still the default grayscale values for the environment. Is there another render pass or anything that needs to be adapted?

EricBoiseLGSVL, 2020-09-25:
@kamkolak

martins-mozeiko, 2020-09-25:
After editing DepthPass file you need to recompile `Shaders/EnvironmentShaders/EnvironmentSimulation.shadergraph` shadergraph. Find it in project window, right click and choose Reimport. This shadergraph compiles shader that is applied to road/buildings/etc and as a separate passes includes depth, lidar & segmentation hlsl file fragments. Note that this shader will use new version only on maps that you build from source or run inside editor from source assets. For prebuilt maps, like BorregasAve this won't change anything, as it includes pre-compiled shaders as platform-specific binaries inside the bundle.

mosshammer, 2020-09-25:
Thanks! That's exactly what I was missing, it's working perfectly fine now

closed 2020-09-25 by mosshammer