lgsvl/simulator #1015: ros2 bridge report "Unknown operation received ..." Thanks

chongxing, 2020-09-25:
I try to run simulator and ros2 bridge (https://github.com/lgsvl/ros2-lgsvl-bridge), what after starting running inside a map from simulator, and ros2 bridge report error, as the following: 

[DEBUG] [rcl]: Initializing ROS client library, for context at address: 0x7fffb6f62eb0
[DEBUG] [rcl]: Initializing node 'bridge' in namespace 'lgsvl'
[DEBUG] [rcl]: Using domain ID of '0'
[DEBUG] [rcl]: Using security: false
[DEBUG] [rcl]: Initializing publisher for topic name 'rosout'
[DEBUG] [rcl]: Expanded topic name '/lgsvl/rosout'
[DEBUG] [rmw_fastrtps_shared_cpp]: Adding topic 'rt/lgsvl/rosout' with type 'rcl_interfaces::msg::dds_::Log_' for node '1.f.bb.51.f4.2b.0.0.1.0.0.0|0.0.1.c1'
[DEBUG] [rcl]: Publisher initialized
[DEBUG] [rcl]: Node initialized
[INFO] [lgsvl-bridge]: Listening on port 9090
[DEBUG] [rmw_fastrtps_shared_cpp]: Adding topic 'xpage_internal_topic' with type 'xpage_internal_discovery' for node '1.f.54.49.e1.26.0.0.3.0.0.0|0.0.1.c1'
[DEBUG] [rmw_fastrtps_shared_cpp]: Adding topic 'xpage_internal_topic' with type 'xpage_internal_discovery' for node '1.f.54.49.e1.26.0.0.3.0.0.0|0.0.1.c1'
[DEBUG] [rmw_fastrtps_shared_cpp]: Adding topic 'xpage_internal_topic' with type 'xpage_internal_discovery' for node '1.f.54.5.36.f.0.0.3.0.0.0|0.0.1.c1'
[DEBUG] [rmw_fastrtps_shared_cpp]: Adding topic 'xpage_internal_topic' with type 'xpage_internal_discovery' for node '1.f.54.5.36.f.0.0.3.0.0.0|0.0.1.c1'
[INFO] [lgsvl-bridge]: Client [127.0.0.1:58440] connected
[DEBUG] [lgsvl-bridge]: Received 186 bytes
[ERROR] [lgsvl-bridge]: Unknown operation received from client (71), disconnecting client
[INFO] [lgsvl-bridge]: Client [127.0.0.1:58442] connected
[DEBUG] [lgsvl-bridge]: Received 186 bytes
[ERROR] [lgsvl-bridge]: Unknown operation received from client (71), disconnecting client
[INFO] [lgsvl-bridge]: Client [127.0.0.1:58444] connected
[DEBUG] [lgsvl-bridge]: Received 186 bytes
[ERROR] [lgsvl-bridge]: Unknown operation received from client (71), disconnecting client
[INFO] [lgsvl-bridge]: Client [127.0.0.1:58446] connected
[DEBUG] [lgsvl-bridge]: Received 186 bytes
[ERROR] [lgsvl-bridge]: Unknown operation received from client (71), disconnecting client

1. simulator is started with Jaguar2015XE (Autoware) car, and bridge address is localhost:9090. (Is it ok to use autoware car with the ros2-lgsvl-bridge?) The running info as the following:
[image]
[image]

2. ros2 bridge is built from  (https://github.com/lgsvl/ros2-lgsvl-bridge). Th running command : ./install/lgsvl_bridge/bin/lgsvl_bridge --port 9090 --log D.  The error message is shown above. 
I try to look search the error message from the source. But still I don't know the cause.

        while (size >= sizeof(uint8_t))
        {
            if (buffer[0] == OP_ADD_SUBSCRIBER)
            {
                handle_add_subscriber();
            }
            else if (buffer[0] == OP_ADD_PUBLISHER)
            {
                handle_add_publisher();
            }
            else if (buffer[0] == OP_PUBLISH)
            {
                handle_publish();
            }
            else
            {
                ERROR("Unknown operation received from client (" << uint32_t(buffer[0]) << "), disconnecting client");
                clients.stop(shared_from_this());
                return;
            }

            if (size == buffer.size())
            {
                break;
            }
            size = buffer.size();
        }

Anything steps I missed or did wrong?  Thanks.

luqiang21, 2020-09-25:
@chongxing Are you running another bridge instance for example inside Apollo? This is typically saying the bridge types are not correct.

martins-mozeiko, 2020-09-25:
I would guess that vehicle selected in simulator is using ROS1 bridge, or simulator version is not from 2020.06 branch - as old ROS2 bridge is using ROS1-like protocol. I'm not sure why master does not point to latest code. New ROS2 bridge was added to code in 2020.06 branch.

`71` is ascii code for `G` - and first thing ROS1 bridge and old ROS2 bridge is doing, is sending string that starts with `GET`

chongxing, 2020-09-25:
@martins-mozeiko I am using Simulator of 2020.06 release binary (lgsvlsimulator-linux64-2020.06).

The vehicle selected is the default Jaguar2015XE (Autoware). Is it the right choice?

chongxing, 2020-09-25:
@luqiang21 Apollo is not running. only running simulator and the bridge.

martins-mozeiko, 2020-09-25:
To know what bridge you are using, check it in web interface. Here's the link to documentation which shows editing dialog of vehicle configuration (which shows bridge type): https://www.lgsvlsimulator.com/docs/vehicles-tab/#how-to-change-the-configuration-of-a-vehicle

chongxing, 2020-09-27:
@martins-mozeiko 

From the webui, the default bridge type is ROS. After changing to ROS2, the ros2 bridge work. 
Thanks very much.

closed 2020-09-27 by chongxing