lgsvl/simulator #1016: How to subscribe canbus topic via ros apollo bridge

ccc3017, 2020-09-25:
Actually, I'm beginner of this field. but Thanks to lgsvl, I'm getting better at this field.
So, This is the issue. Now, I ran the apollo 3.0 with lgsvl as you mentioned on your documentation[https://www.lgsvlsimulator.com/docs/apollo-instructions/], Of course It worked.
And then, I wanted to see canbus topic publishing on my rqt window and subscribe it. but It said "can not get message class for type "pb_msgs/Chassis".
[image]
How can I subscribe this. I need help....

EricBoiseLGSVL, 2020-09-25:
@rongguodong Do you know what could be the issue?

rongguodong, 2020-09-25:
ROS-based Apollo 3.0 is very old. I am not sure if it still works well with latest version of simulator. Is it possible to use our fork of Apollo 5.0 or even Apollo master (both are CyberRT-based)?

@hadiTab @daviduhm Do you have any ideas on this ROS message type issue?

hadiTab, 2020-09-25:
@ccc3017 apollo 3.0 uses a modified version of ROS that serializes to protocol buffers. Standard ROS cannot handle these messages. The rqt tools are all part of standard ROS and I don't think Apollo has modified any of those tools to be able to parse the protobuf messages.

Basically, you won't be able to use any of those tools on any topics that don't have standard ROS types. I believe the only standard ROS topics are the camera images and the lidar pointcloud.

ccc3017, 2020-09-26:
@rongguodong 
I will try it with other version of apollo as you said.
Anyway, You mean it should be possible to subscribe canbus topic with other version of apollo?

@hadiTab 
Hmm... That makes sense. Thank you.
I'm using ubuntu 16.04.
Then, Is there any way that I can subscribe Canbus topic from simulator?

lemketron, 2020-09-26:
What @hadiTab was saying is that the rqt (ROS) tools won't work with Apollo 3.0 (nor with 3.5 or newer, which use CyberRT instead of ROS). You can use CyberMonitor to view the messages sent to Apollo 3.5 or newer, and you can investigate other CyberRT tools at the Apollo github. 

Check the following links:
https://github.com/ApolloAuto/apollo/tree/master/cyber#documents
https://github.com/ApolloAuto/apollo/blob/master/docs/cyber/CyberRT_Developer_Tools.md

ccc3017, 2020-09-26:
@lemketron 
Oh... 
Then I need to run newer version of apollo and CyberRT bridge to use CyberMonitor.
I will try it and come back Thanks!

ccc3017, 2020-09-27:
@lemketron 
A new issue came up... I ran apollo 5.0 using CyberRT bridge. It worked well.
And As you mentioned above, I tried to use CyberRT monitor.
But I got this message
[image]
How can I fix it and run the CyberRT monitor?

closed 2020-12-02 by EricBoiseLGSVL