lgsvl/simulator #1018: Changing EV span position.

Mojojojooo, 2020-09-27:
Can we make the Ego vehicle span at a different location in the map just by making changes somewhere in the code without using unity or python API as my work involves changing the position frequently.

Please Help me here.

ntutangyun, 2020-09-28:
hi, if the position is frequently changing, I believe you're better off using API instead of Hard coding the positions in the source code

Mojojojooo, 2020-09-28:
For me, it is like I can either use API or my ROS node. So I wanted to make changes in the source code itself.

EricBoiseLGSVL, 2020-09-28:
You can use the API like @ntutangyun suggested and I also recommend.  You can write a loop that cycles the positions you want.  You can also raycast down to make sure the position is on the map before placing.  If you really need the hard coded places, you can edit the map and place a bunch of SpawnInfo objects and write some code to cycle these positions.

closed 2020-12-02 by EricBoiseLGSVL