lgsvl/simulator #1020: Syncronization of NPC

asifsid-32, 2020-09-28:
Hello LGSVL team,

I have been using lgsvl simulator for a while now. I want to sync my ego vehicle movement and NPC. Is there any way we can do this by using python api's since changes in source code looks too complex and will need some time also.

If we are able to do it using some python api that will be a great help.

EricBoiseLGSVL, 2020-09-28:
@asifsid-32 What exactly are you trying to do?  Why do you need to sync NPC?  Python API changes are not too bad.  Just update commands in Simulator and update Python API code.  Then build a new binary.  You can test in Editor from loading scene while you develop.

asifsid-32, 2020-09-28:
> @asifsid-32 What exactly are you trying to do?  Why do you need to sync NPC?  Python API changes are not too bad.  Just update commands in Simulator and update Python API code.  Then build a new binary.  You can test in Editor from loading scene while you develop.

I want my NPC to move at certain point of time. So I need to sync it with my ego vehicle. As soon as my ego vehicle is in certain range I want the Npc to move. 

Now can this be done with the available API's ??

lemketron, 2020-09-29:
> I want my NPC to move at certain point of time. So I need to sync it with my ego vehicle. As soon as my ego vehicle is in certain range(distance), I want the Npc to move.
> 
> Now can this be done with the available API's ??

There's a great example of pretty much exactly what you want to do in quickstart script 26:
https://github.com/lgsvl/PythonAPI/blob/master/quickstart/26-npc-trigger-waypoints.py

lemketron, 2020-09-29:
@asifsid-32 In addition, you can check out the [NPC python API documentation](https://www.lgsvlsimulator.com/docs/python-api/#npc-vehicles):

> The trigger distance, if used, provides a method to pause the NPC until an ego vehicle approaches. The NPC will begin to drive as soon as its distance to an ego vehicle drops below the value specified as trigger distance in the DriveWaypoint. You can receive information on progress by setting the on_waypoint_reached callback. (See related example in docs.)

EricBoiseLGSVL, 2020-09-29:
Thanks @lemketron .  Yep, npc trigger waypoint is what you want.

closed 2020-12-02 by EricBoiseLGSVL