lgsvl/simulator #1022: Reading Apollo control message on Windows

eyadshak, 2020-09-28:
How can I read Apollo control messages (/apollo/control) on a Windows machine? I tried to reach messages coming from port 9090 but I didn't get far with that approach. What format are they being sent?

ntutangyun, 2020-09-28:
May I know your hardware and software stack here. 

Is Apollo running on another ubuntu machine ? and this machine is connected with the windows machine in a local network?

eyadshak, 2020-09-28:
I am running Apollo 5.0 on an Ubuntu 18.04 machine, and running the latest LGSVL , built from source, on a separate Windows 10 machine. Both have a Quadro p4000 GPU.

All functionalities are working fine, including perception. The car can navigate from point A to B autonomously just fine.
Machines are on the same local network.

I want to record the control messages reaching LGSVL on the Windows 10 machine (published by Apollo on /apollo/control/).

ntutangyun, 2020-09-28:
hi @eyadshak  basically LGSVL communicates through a socket connection with the Apollo cyber bridge module. 

The cyber bridge module supports multiple clients.

So you could write a script, set up a socket connection to the same bridge address (host and port) and register a reader for the `/apollo/control` channel. 

You may refer to [this script](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Cyber/CyberBridge.cs) on the protocols of cyber bridge.

eyadshak, 2020-09-28:
@ntutangyun thank you very much, I'll have a look at the script.

closed 2020-09-28 by eyadshak