lgsvl/simulator #1024: 'follow_closest_lane' can not set speed in some sceanario

qwetqwe, 2020-09-29:
Hi team,
I want to make npc follow the lane in a 1 m/s speed. At first, it works well. After the npc pass first traffic light, it speeds up to a high speed. I don't know why.
```
import os
import lgsvl
import math
sim = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "192.168.1.99"), 8181)
if sim.current_scene == "SanFrancisco":# SanFrancisco AutonomouStuff BorregasAve
  sim.reset()
else:
  sim.load("SanFrancisco")
spawns = sim.get_spawn()
state = lgsvl.AgentState()
state.transform = spawns[0]
forward = lgsvl.utils.transform_to_forward(spawns[0])
state.velocity = 0 * forward
a = sim.add_agent("vehicle_Lincoln2017MKZ", lgsvl.AgentType.EGO, state)
a.connect_bridge("192.168.1.102", 9090)
state.velocity = 0 * forward
up = lgsvl.utils.transform_to_up(state.transform)
layer_mask = 0
for bit in [0, 10, 11, 12]: # do not put 9 here, to not hit EGO vehicle itself
  layer_mask |= 1 << bit
hit = sim.raycast(state.transform.position+30*forward, -up, layer_mask)
state.transform.position = hit.point
npc1 = sim.add_agent("Sedan", lgsvl.AgentType.NPC, state)
npc1.follow_closest_lane(True, 1) #follow
controllables = sim.get_controllables("signal")
control_policy = "trigger=100;green=60;yellow=2;red=1;loop"
for c in controllables:
  c.control(control_policy)
sim.run(time_limit = 0)
```

EricBoiseLGSVL, 2020-09-29:
@qwetqwe This might be a bug.  We had a refactor of NPCs and there may be a regression.  We will look into it.  If you have video and any other info on this, it would be great, thanks.

ntutangyun, 2020-09-29:
Hi @qwetqwe @EricBoiseLGSVL ,

below is my understanding:

when the `follow_cloest_lane` API is called, it calls the function `SetFollowCloestLane()` in file `NPCLaneFollowController.cs`, where the speed is passed and assigned to `normalSpeed`.
That's why it works initially. 

But when NPC reached the end of its traffic lane and computing its next lane, this `normalSpeed` gets overridden as below:

```c#
// NPCLaneFollowController.cs
...
protected void GetNextLane() 
{
    ...
    aggressionAdjustRate = laneSpeedLimit / 11.176f; // 11.176 m/s corresponds to 25 mph
    normalSpeed = RandomGenerator.NextFloat(laneSpeedLimit - 3 + aggression, laneSpeedLimit + 1 + aggression);
    ...
}
....
```
which is set to about `laneSpeedLimit`. that's why it speeds up to a high speed.

I used an additional variable here to record the user-defined speed from the API call. 

```c#
// NPCLaneFollowController.cs
...
protected void GetNextLane() 
{
    ...
    if (customSpeed > 0) {
        normalSpeed = Mathf.Min(customSpeed, laneSpeedLimit);
    } else {
        aggressionAdjustRate = laneSpeedLimit / 11.176f; // 11.176 m/s corresponds to 25 mph
        normalSpeed = RandomGenerator.NextFloat(laneSpeedLimit - 3 + aggression, laneSpeedLimit + 1 + aggression);
    }
    ...
}
....
```

EricBoiseLGSVL, 2020-09-29:
Thanks for the find, we will look at this asap

qwetqwe, 2020-09-30:
@ntutangyun  
It seems right, do I need to build by unity again after modified?

ntutangyun, 2020-09-30:
> @ntutangyun  
> It seems right, do I need to build by unity again after modified?

Yes. Please try out first with Unity Editor. If it works the way you want. You could build and run the executable file alone.

EricBoiseLGSVL, 2020-09-30:
@ntutangyun Thanks for the help and fix.  We have added this as a task to fix asap.  Thanks @qwetqwe for the bug report.

EricBoiseLGSVL, 2020-10-06:
@ntutangyun @qwetqwe Thanks again, this will be in the 07 release.

closed 2020-10-06 by EricBoiseLGSVL