lgsvl/simulator #1025: Editing Map in unity editor

asifsid-32, 2020-09-29:
Hello Team,

I am facing issues in loading up the screen in unity editor. I have followed all the points from the given link https://www.lgsvlsimulator.com/docs/add-new-map/#import-assets. When i run the scene in unity editor i get a blank screen in which my ego car falls from the sky. 

Please find attached screenshot of the same. 
[image]

asifsid-32, 2020-09-29:
> Hello Team,
>
> I am facing issues in loading up the screen in unity editor. I have followed all the points from the given link https://www.lgsvlsimulator.com/docs/add-new-map/#import-assets. When i [...]

Followings issues i faced when i checked it - 
[image]

EricBoiseLGSVL, 2020-09-29:
Don't worry about check, that is more of an internal check for assets.  Be sure to use git lfs pull to make sure you have assets.  You can check that textures are not URLs but actual .png files.

asifsid-32, 2020-09-29:
> Don't worry about check, that is more of an internal check for assets.  Be sure to use git lfs pull to make sure you have assets.  You can check that textures are not URLs but actual .png files.

@EricBoiseLGSVL about the ego vehicle falling from the sky that is the issue ? Also what has to be done to load the actual map in the scene to edit it.

EricBoiseLGSVL, 2020-09-29:
If you do not have git lfs installed when cloning the repo, you will not have mesh assets.  If you don't have textures (you won't see anything) and if you don't have meshes (you won't have collider data) and the vehicle will fall.  It's odd you have the vehicle assets and not the map.  Did you edit the map at all?

asifsid-32, 2020-09-29:
> If you do not have git lfs installed when cloning the repo, you will not have mesh assets.  If you don't have textures (you won't see anything) and if you don't have meshes (you won't have collider [...]

I installed git lfs. No I am trying to load the scene first. Then I have to add calib patterns to it. I downloaded the single lane road map and added it to the path external/environments/singleroadmap 

Not able to find out the issue here !! Can you please help ?

EricBoiseLGSVL, 2020-09-29:
@asifsid-32 Can you please screen shot what the map looks like without running it so I can see the spawnInfo position and the collider on the road.

asifsid-32, 2020-09-29:
> @asifsid-32 Can you please screen shot what the map looks like without running it so I can see the spawnInfo position and the collider on the road.

PFA screenshots for the same .... 
[image]
[image]
[image]

EricBoiseLGSVL, 2020-09-29:
Can you please open the Scene tab like so
[image]

asifsid-32, 2020-09-29:
> Can you please open the Scene tab like so
> [image]
> 
> Game tab is only for when Editor is playing. Since simulator spawns scene cameras on play, there is no image in this tab when not playing. You also need to start simulation from loader scene, not the map scene. You build the map bundle and add path to map path in WebUI. Developer settings are still WIP and may have issues.
@EricBoiseLGSVL there's no tab as Scene, just the game tab i am able to find. Also i was able to start the loaderscene but when i run it stops at the "API Ready" page.

PFA ss from the unity 
[image]
[image]

asifsid-32, 2020-09-29:
> Can you please open the Scene tab like so
> [image]
>
> Game tab is only for when Editor is playing. Since simulator spawns scene cameras on play, there is no image in this tab when not playing. You also [...]

Just Got the scene tab .. 
[image]

asifsid-32, 2020-09-29:
Following error i am getting when i opened the .png file present in the materials folder ....
So does this means that lfs wasn't installed properly ? Do i have to install it again ?
[image]

martins-mozeiko, 2020-09-29:
Run `git lfs ls-files` in folder where you cloned the repository. It will print out status of all files under LFS. Each file **MUST** have an asterix (`*`) next to it. If you see only dash (`-`) that means git lfs files are not downloaded. In such case run `git lfs pull` and that will fix your repository.

For more information on git lfs see documentation here: https://www.atlassian.com/git/tutorials/git-lfs

asifsid-32, 2020-09-30:
> I installed git lfs. No I am trying to load the scene first. Then I have to add calib patterns to it. I downloaded the single lane road map and added it to the path [...]

@martins-mozeiko do I have to run from the loader scene or I can directly edit the map ??

asifsid-32, 2020-09-30:
> Run `git lfs ls-files` in folder where you cloned the repository. It will print out status of all files under LFS. Each file **MUST** have an asterix (`*`) next to it. If you see only dash (`-`) [...]

@EricBoiseLGSVL @martins-mozeiko Thanks for the support i am able to load the map in the unity editor. Now i want to place calibration patterns here. But in general i was using https://content.lgsvlsimulator.com/vehicles/lexusrx2016/ . So is it also open source or do i have to use Jaguar2015XE only. Also @EricBoiseLGSVL a little support on how to place calib patterns and reflect them back on simulator will prove a great help. 
[image]

asifsid-32, 2020-09-30:
Hii @EricBoiseLGSVL , @martins-mozeiko 

I am not able to generate asset bundle file after making changes and building the simulator. Also when i run the loader screen it directly gets to API ready instead of open browser. 

PFA screenshot for the same 
[image]

EricBoiseLGSVL, 2020-09-30:
Did you build the new CubeTown asset?  Did you include windows support in the installation of Unity 2019.3.15f1?  You can do this in Unity Hub.  What errors do you get when you try to make a build?  Please look up how to get Editor logs in Unity.

Jaguar2015XE is our only open source vehicle but you can create private assets for vehicles.  It is up to you.  If you want to make a new vehicle, please see our tutorial on this [here](https://youtu.be/R_Z65kimCII).

I'm not sure on camera calibration but you could add the vehicle model into a scene, add a camera with the settings needed in the editor as a child of the vehicle as well as the calibration images on planes.  Then align the camera view to the real vehicle and save the transform data of the camera.  Then apply the position/rot of the camera to the sensor parameters in the simulation setup in the WebUI.

asifsid-32, 2020-09-30:
> Did you build the new CubeTown asset? Did you include windows support in the installation of Unity 2019.3.15f1? You can do this in Unity Hub. What errors do you get when you try to make a build? [...]

@EricBoiseLGSVL is there a difference between building the simulator and building the CubeTown asset ? how do we have to build the CubeTown asset ?

Yes windows support is included in the istallation of unity 2019.3.15f1. When i build there were no errors and it got build successfully . 

But i am not able to find the environment_CubeTown file in assetbundle folder....

Any help will be appreciable ... Thanks in advance !!

EricBoiseLGSVL, 2020-09-30:
Is this a duplicate of #1034?  Please close the issue that is not needed

closed 2020-09-30 by asifsid-32