lgsvl/simulator #1026: Information related to WalkWaypoint Function

rajesh557, 2020-09-29:
Hello LGSVL team,

i wanted NPC to stop at a certain defined distance from the EGO vehicle, and NPC should move again when its distance from ego vehicle drop below defined distance.
I am using 33-wait-for-distance trigger code(modified below highlighted part)
I was using **wp = lgsvl.WalkWaypoint(position=hit.point, speed=speed, idle=0, trigger_distance=0, trigger=None)** this function to achieve what i mentioned above.
Can anyone please help me understand what trigger_distance does, and how can i achieve my requirement.

EricBoiseLGSVL, 2020-09-29:
@rajesh557 trigger distance is in meters from npc to ego.  NPC waits until Ego is within the distance set, then moves.

lemketron, 2020-09-29:
I think you meant [31-wait-for-distance-trigger.py](https://github.com/lgsvl/PythonAPI/blob/master/quickstart/31-wait-for-distance-trigger.py).

lemketron, 2020-09-29:
@rajesh557  In addition, you can check out the [NPC python API documentation](https://www.lgsvlsimulator.com/docs/python-api/#npc-vehicles):

> The trigger distance, if used, provides a method to pause the NPC until an ego vehicle approaches. The NPC will begin to drive as soon as its distance to an ego vehicle drops below the value specified as trigger distance in the DriveWaypoint. You can receive information on progress by setting the on_waypoint_reached callback. (See related example in docs.)

rajesh557, 2020-09-30:
> @rajesh557 trigger distance is in meters from npc to ego. NPC waits until Ego is within the distance set, then moves.

The trigger distance, if used, provides a method to pause the NPC until an ego vehicle approaches. The NPC will begin to drive as soon as its distance to an ego vehicle drops below the value specified as trigger distance in the DriveWaypoint. You can receive information on progress by setting the on_waypoint_reached callback. 

here in the documentation it is mentioned that NPC will pause, where will npc pause? is that specified by us?

my problem is to control moment of NPC depending on the ego vehicle. NPC should be moving randomly and when it comes in some distance with ego vehicle, it should pause, and when ego is out of the specified distance it should move again, how to achieve this?
sample code for this will help a lot as i am not much well versed in this.
Thanks in advance

EricBoiseLGSVL, 2020-09-30:
It will pause at the waypoint you set the trigger, e.g. 31-wait-for-distance-trigger.py
```
wp = lgsvl.DriveWaypoint(position=hit.point, speed=speed, angle=angle, idle=0, trigger_distance=0, trigger = trigger)
```

Again, this is the opposite of the trigger logic that is present.  You will need to alter the code to do this in NPCWaypointBehaviour.cs.  Here it returns null when the distance to EGO is greater than the trigger distance given.  It will need another bool value for trigger data and API changes to pass to the NPC logic or create a new TriggerIE to set this logic.
```
private IEnumerator NPCTriggerIE()
{
    while(SimulatorManager.Instance.AgentManager.GetDistanceToActiveAgent(transform.position) > LaneTriggerDistance[CurrentIndex])
    {
         yield return null;
    }
    TriggerCoroutine = null;
}
```

rajesh557, 2020-10-01:
@EricBoiseLGSVL can i use hit = sim.raycast() in autonomous stuff map?

EricBoiseLGSVL, 2020-10-01:
Sure! Any map.

rajesh557, 2020-10-01:
But it gives an error if I use this in autonomous stuff map.
I get error in that line as it has "no attribute point'

EricBoiseLGSVL, 2020-10-01:
What is the point that you are trying to raycast?  Is it on the map?