lgsvl/simulator #1028: [NPCController] built-in collision avoidance is not working while turning in Junction

ntutangyun, 2020-09-29:
Hi,

I encountered this weird scenario where NPC's built-in collision avoidance is not working while turning in junctions.

Here is a [video](https://drive.google.com/file/d/1DZh_k98svttExs-M2QMLJY-C6OcOTpw_/view?usp=sharing)

The NPCs are controlled by the slightly modified version of `NPCLaneFollowController.cs`.

I guess this behavior is caused by the following piece of code:

```C#
    protected float SetFrontDetectSpeed()
    {
        var blocking = frontClosestHitInfo.transform;
        blocking = blocking ?? rightClosestHitInfo.transform;
        blocking = blocking ?? leftClosestHitInfo.transform;

        float tempS = 0f;
        if (Vector3.Dot(transform.forward, blocking.transform.forward) > 0.7f) // detected is on similar vector
        {
            if (frontClosestHitInfo.distance > stopHitDistance)
            {
                tempS = (normalSpeed) * (frontClosestHitInfo.distance / stopHitDistance);
            }
        }
        else if (Vector3.Dot(transform.forward, blocking.transform.forward) < -0.2f && (isRightTurn || isLeftTurn))
        {
            // THIS CONDITION IS TRIGGERED
            tempS = normalSpeed;
        }
        return tempS;
    }
```

EricBoiseLGSVL, 2020-09-29:
@ntutangyun yes we have some issues from a refactor.  We are looking at solutions asap. Thanks for the video

ntutangyun, 2020-09-29:
I used this temporary modification. But not sure what side effects it introduces.

```c#
        if (frontClosestHitInfo.distance > stopHitDistance)
        {
            tempS = (normalSpeed) * (frontClosestHitInfo.distance / stopHitDistance);
        }

        // if (Vector3.Dot(transform.forward, blocking.transform.forward) > 0.7f) // detected is on similar vector
        // {
        //     if (frontClosestHitInfo.distance > stopHitDistance)
        //     {
        //         tempS = (normalSpeed) * (frontClosestHitInfo.distance / stopHitDistance);
        //     }
        // }
        // else if (Vector3.Dot(transform.forward, blocking.transform.forward) < -0.2f && (isRightTurn || isLeftTurn))
        // {
        //     Debug.Log("this condition is triggered");
        //     tempS = normalSpeed;
        // }
```

EricBoiseLGSVL, 2020-09-29:
It will block no matter the vector of the blocking transform.  Should be ok but will have issues in intersections.  Still better than hitting EGO :)

EricBoiseLGSVL, 2020-09-30:
@ntutangyun Thanks for the bug find, we have added this to fix asap.

closed 2020-10-06 by EricBoiseLGSVL