lgsvl/simulator #1029: How to load autoware_msgs?

Amina88, 2020-09-29:
Hi guys, Im having issue publishing the data from topics for example vehicle cmd topic? $ rostopic echo /vehicle_cmd ERROR: Cannot load message class for [autoware_msgs/VehicleCmd]. Are your messages built? Any one has an ideas how to resolve this plzzzz. thank you
[image]

EricBoiseLGSVL, 2020-09-29:
Any idea @cyberphysicalrobotics ?

EricBoiseLGSVL, 2020-10-20:
@hadiTab adding you to ping @cyberphysicalrobotics

hadiTab, 2020-10-20:
@Amina88 you can install autoware_msgs on your machine using `sudo apt install ros-$ROS_DISTRO-autoware-msgs`.
In general, if you are running Autoware inside a docker container and you want to inspect messages or launch a ros tool it is best to do that inside the same docker container.

closed 2020-12-02 by EricBoiseLGSVL