lgsvl/simulator #1030: Autoware path plan failed

jiameixie, 2020-09-30:
I followed the instructions in https://www.lgsvlsimulator.com/docs/autoware-instructions/.   
The car doesn't move after I click 2D Pose Estimate and 2D Nav Goal.

1. The Vehicle mode and point raw status are error
2. All the plans are failed
Can anyone help me with it? Thanks.

[image]
[image]
[image]
[image]

EricBoiseLGSVL, 2020-09-30:
@cyberphysicalrobotics do you know what is causing the issue?

EricBoiseLGSVL, 2020-10-20:
@hadiTab adding you to ping @cyberphysicalrobotics

hadiTab, 2020-10-20:
@jiameixie can you show what point on the map you are selecting as the destination? That error usually shows up if the destination you chose is not valid (i.e. Autoware can't find a path to the point when searching through the vector map).

Also, an annoying issue with the planner in Autoware is that if you send an invalid destination request you need to restart the node to try again, otherwise all subsequent requests also fail. So if you select a destination and you get that error message and want to try again, go back to the `Quick Start` tab in the runtime manager and deactivate and reactivate `Mission Planner`.

jiameixie, 2020-10-21:
@hadiTab Thank you for explaining the relevant knowledge to me. I forget what direction I chose. Maybe it was invalid.

closed 2020-10-21 by jiameixie