lgsvl/simulator #1031: Problems with traffic light detection

ucmar17, 2020-09-30:
When I am trying to detect traffic light in the Cubetown simulator, it keeps detecting the yellow and red lights as green light, and I am not sure why Apollo does that. The simulator version I am running is 2020.06 and building it using Unity 2019.3.15f1. The Apollo version I am using is the lgsvl-apollo-5.0 fork which I built from source.

EricBoiseLGSVL, 2020-09-30:
@rongguodong Any idea why this is happening?

EricBoiseLGSVL, 2020-10-06:
@ucmar17 It's because we don't have signal ids in cubetown.  We will be updating this in the next release.  This will happen automatically on HDmap import and existing maps will be updated. Right @rongguodong and @luqiang21 ?

lemketron, 2020-10-06:
> The Apollo version I am using is the lgsvl-apollo-5.0 fork which I built from source.

From what commit did you build Apollo, and are you building from the "simulator" branch?

Also, are you sure you have the latest 2020.06 environment for CubeTown?

The signal IDs should only matter if you're trying to use the new [Modular Testing](https://www.lgsvlsimulator.com/docs/modular-testing/) feature (which uses ground truth from simulator instead of running Apollo perception and traffic light modules, in which case the signal IDs in the simulator environment need to match those in the Apollo map).

If you have not tried this, I would encourage you to take a look since it makes it much easier to run Apollo and Simulator on a single machine by publishing the necessary traffic signal information directly on the `/apollo/perception/traffic_light` topic.

If traffic signal perception (using camera sensors in Apollo) is not working then there may be some other problem (as happens from time to time; this can be caused by problems in the environment, or issues with Apollo perception as has happened in Apollo master).

ucmar17, 2020-10-07:
Hi @lemketron I built Apollo using the commit number 666007b82ba40857677650bdbbe929a6e2fb41ef, and yes I did build from the simulator branch. I am also sure that I am using the "release-2020.06" branch from the simulator Github to build the simulator.

I am also running the simulator and Apollo on the same machine and I am pretty sure the traffic signal information is being published to /apollo/perception/traffic_light, as I can see it in cyber_monitor. I am still not sure what other problems could be taking place. Do you have any other suggestions?

EricBoiseLGSVL, 2020-10-08:
@ucmar17 If you are using Apollo and Simulator on the same machine you maybe running out of memory to run traffic light detection module in Apollo. Odd that it is working on Borregas and not CubeTown

lemketron, 2020-10-08:
@ucmar17 Are you using the "modular testing" feature where simulator publishes the signal information to /apollo/perception/traffic_light? If not, try it (and leave traffic light module off in dreamview). 

If this doesn't work then it could be a problem with signal ID mismatch between the Apollo map and the signal ID in the environment, but that should be fixed in the latest map and environment.

If you need to run traffic light perception then you should run "watch -n1 nvidia-smi" to make sure traffic light perception is loading successfully in the GPU running and not crashing. You can also check the Apollo logs for traffic light and see if there is any helpful info in there.

ucmar17, 2020-10-08:
@lemketron I am not using the modular testing feature, as my project requires me to use traffic light detection through Apollo. However, I just tried it with modular testing (to test if that is at least working) and the traffic light detection is working in Cubetown as well as Borregas Ave. This means that it is probably not an issue with the signal ID mismatch.

Also, @EricBoiseLGSVL I did also run "watch -n1 nvidia-smi" to make sure that the GPU is running and not crashing, which seems to be the case. Even the Apollo logs seem to be fine and nothing is in there for the traffic light module. 

I am not sure what is happening. Would any of you happen to have any other suggestions?

EricBoiseLGSVL, 2020-10-09:
@ucmar17 can you test the light detection during the evening, change time of day?  Maybe the angle of the lights to the sun is causing an issue in CubeTown?

ucmar17, 2020-10-09:
@EricBoiseLGSVL Yep, that worked! Thank you so much for your help. Do you think the sunlight issue will be fixed in the next release of the simulator? For now, I can continue testing during the night time, but it would be helpful for me to be able to test during daylight as well.

EricBoiseLGSVL, 2020-10-12:
@ucmar17 We are currently working on adding multiple traffic light types and updating the logic so emission works better.  It will be in the next release or the one after.  If you want to fix now, you can turn off specular in the traffic light material and increase the emission values in the code.  It should help until we can release the new lights