lgsvl/simulator #1032: Help in creating a scenario

rajesh557, 2020-09-30:
Hello team, i wanted to create a scenario where a pedestrian is moving in some waypoint defined, and it should pause when EGO vehicle comes close to it, and when its distance from ego vehicle falls below defined distance it should start moving again. 

`#!/usr/bin/env python3
#
# Copyright (c) 2019 LG Electronics, Inc.
#
# This software contains code licensed as described in LICENSE.
#

import os
import lgsvl
import random

sim = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "127.0.0.1"), 8181)
if sim.current_scene == "AutonomouStuff":
  sim.reset()
else:
  sim.load("AutonomouStuff")

spawns = sim.get_spawn()

state = lgsvl.AgentState()
state.transform = spawns[0]
a = sim.add_agent("Lincoln2017MKZ (Apollo 5.0)", lgsvl.AgentType.EGO, state)
sx = spawns[0].position.x
sy = spawns[0].position.y
sz = spawns[0].position.z

layer_mask = 0
layer_mask |= 1 << 0  # 0 is the layer for the road (default)

state.transform.position =  lgsvl.Vector(48.4891510009766, -2.56685018539429, -66.2934951782227)
pedestrian_position = lgsvl.Vector(48.4891510009766, -2.56685018539429, -66.2934951782227)
state.transform.rotation =  lgsvl.Vector(1.61875581741333, 134.981918334961, 359.785827636719)
pedestrian = sim.add_agent("Pamela", lgsvl.AgentType.PEDESTRIAN, state)

agents = {
    a: "EGO",
    pedestrian: "Bob"
}

# Executed upon receiving collision callback -- pedestrian is expected to walk into colliding objects
def on_collision(agent1, agent2, contact):
    name1 = agents[agent1]
    name2 = agents[agent2] if agent2 is not None else "OBSTACLE"
    print("{} collided with {}".format(name1, name2))

a.on_collision(on_collision)
pedestrian.on_collision(on_collision)

waypoints = []

waypoints = [
lgsvl.WalkWaypoint(lgsvl.Vector(26.0313243865967, -2.78846454620361, -84.4243392944336), speed=1, idle=0, trigger_distance=5, trigger=None)]
'''
effector = lgsvl.TriggerEffector("TimeToCollision", {})
trigger = lgsvl.WaypointTrigger([effector])
wp = lgsvl.WalkWaypoint(position=pedestrian_position, speed=2.2, idle=0, trigger_distance=10, trigger=None)'''

def on_waypoint(agent, index):
    print("waypoint {} reached".format(index))

def agents_traversed_waypoints():
    print("All agents traversed their waypoints.")
    sim.stop()

# The above function needs to be added to the list of callbacks for the pedestrian
pedestrian.on_waypoint_reached(on_waypoint)
sim.agents_traversed_waypoints(agents_traversed_waypoints)

# The pedestrian needs to be given the list of waypoints. A bool can be passed as the 2nd argument that controls
# whether or not the pedestrian loops over the waypoints (default false)
pedestrian.follow(waypoints, False)

input("Press Enter to run")

sim.run()

`
This is the code i am using from making the pedestrian pause, and again continue when its distance from ego vehicle is less than the defined distance in walkwaypoint.
But the pedestrian does not pause and it collides with EGO vehicle, am i missing something here?
and what changes are needed to rectify this and create required scenario.
Thank you in advance 👍

EricBoiseLGSVL, 2020-09-30:
@rajesh557 "when its distance from ego vehicle falls below defined distance"  Are you saying that you want the pedestrian to continue the path when the EGO distance to the pedestrian is greater than the trigger distance specified?

rajesh557, 2020-09-30:
> @rajesh557 "when its distance from ego vehicle falls below defined distance" Are you saying that you want the pedestrian to continue the path when the EGO distance to the pedestrian is greater than the trigger distance specified?

hey i just want pedestrian to stop moving, when pedestrian's distance from ego vehicle is less than trigger distance. rest of the time pedestrian should continue its path

EricBoiseLGSVL, 2020-09-30:
The current behavior for wait for distance stops agent until <= distance.  It seems you are asking for the opposite behavior.  You will need to edit the PythonAPI and Simulator code for the wait for distance and invert the logic.  You could also try to add a bool flag to wait for distance logic to invert the logic.

rajesh557, 2020-09-30:
@EricBoiseLGSVL where can i find the simulator code for wait for distance?

EricBoiseLGSVL, 2020-09-30:
#1026 has the example of the simulator code.

asifsid-32, 2020-10-07:
Hii @EricBoiseLGSVL if we make following changes in the simulator source code and build it --- 

 private IEnumerator NPCTriggerIE()
    {
        while(SimulatorManager.Instance.AgentManager.GetDistanceToActiveAgent(transform.position) < LaneTriggerDistance[CurrentIndex])
        {
            yield return null;
        }
        TriggerCoroutine = null;
    }

Will it work ?
Also what necessary changes will be required in PythonApi ?

EricBoiseLGSVL, 2020-10-07:
@asifsid-32 Sure that will work, just remember to change when you pull next release.  I will see if we can support this behavior in the release after 07.

asifsid-32, 2020-10-08:
> @asifsid-32 Sure that will work, just remember to change when you pull next release. I will see if we can support this behavior in the release after 07.

@EricBoiseLGSVL  Do we have to change anything in the PythonAPI also ?

EricBoiseLGSVL, 2020-10-08:
I don't think so because you are changing the logic in simulator from values sent from PythonAPI.