lgsvl/simulator #1033: Information related to TimeToCollision and WaitForDistance

rajesh557, 2020-09-30:
Hey team, can anyone explain what is the use if TimeToCollision and WaitForDistance. and where can i find its documentation.
And is there any way that i can make NPC or pedestrian to stop or move depending on its distance from EGO vehicle?

EricBoiseLGSVL, 2020-09-30:
@rajesh557 Time to collision sensor takes the current velocity of the EGO and the npcs velocity and calculates the time to collision.  Wait for Distance will pause an agent until the distance to the EGO is <= agent and continues the pathing.

rajesh557, 2020-09-30:
@EricBoiseLGSVL 
`parameters = {"max_distance": 4.0}
    effector = lgsvl.TriggerEffector("WaitForDistance", parameters)
    trigger = lgsvl.WaypointTrigger([effector])
  wp = lgsvl.DriveWaypoint(position=hit.point, speed=speed, angle=angle, idle=0, trigger_distance=0, trigger = trigger)
  waypoints.append(wp)`

i am referring to 31-wait-for-distance.py code.
in the above code parameters = {"max_distance": 4.0}
what does max_distance mean:4 mean?

rajesh557, 2020-09-30:
@EricBoiseLGSVL sorry but i didnt understand Time to collision and wait for distance, can you please elaborate? or explain it with an example

EricBoiseLGSVL, 2020-09-30:
@rajesh557 No worries, TimeToCollision is a sensor that outputs the pos/neg time to collision with NPC vehicles.  Wait for distance is an agent API trigger that pauses the NPC at a waypoint specified until the EGO is within the distance specified.
e.g. Taking current EGO velocity it will collide in 5 sec with NPC[0], this is sent to bridge as a msg.
e.g. NPC sedan waypoint index 3 is set to have a trigger wait for distance of 4.  When the NPC reaches waypoint 3 it checks that it needs to wait until EGO is within 4 meters before moving to the next waypoint.  Once it does, it moves and evaluates the next waypoint, if not, it stays at waypoint 3.

closed 2020-10-20 by EricBoiseLGSVL