lgsvl/simulator #1034: Editing CubeTown Map in unity editor

asifsid-32, 2020-09-30:
Hello LGSVL team,

 Can anyone from the team please explain me how to get the mentioned file in the screen shot after building the code 

Filename - "enivronment_Moderncity" 
also what is the file format ?
[image]

EricBoiseLGSVL, 2020-09-30:
@asifsid-32 file format is AssetBundle. This is a unity specific file format for engine assets.  You create a bundle from the Simulator -> Build...  Here you select the Assets/External/Environments/YourMap to build.  Make sure you don't build the binary at the same time or it takes a while.  Be aware that the first build will take a bit but following ones will be faster.  Also, be sure to have both windows and linux support because bundles are meant to be cross platform.

[image]

asifsid-32, 2020-09-30:
> @asifsid-32 file format is AssetBundle. This is a unity specific file format for engine assets.  You create a bundle from the Simulator -> Build...  Here you select the Assets/External/Environments/YourMap to build.  These are in the root of the source in the AssetBundles folder
> 
> [image]
> 
> 

Sorry @EricBoiseLGSVL I didn't get you !! When we build the simulator we only have option for path where executable will build.. 

When I built the simulator I am not able to find environment_cubetown inside asset bundle.. 

What can be the issue here ??

asifsid-32, 2020-09-30:
When i built the simulator from unity i get an asset bundle  folder with following files- 

[image]

Also i got some files extracted from environment_CubeTown - 

[image]

Any thing i am doing wrong ?

asifsid-32, 2020-09-30:
@EricBoiseLGSVL any suggestions ?

asifsid-32, 2020-09-30:
@EricBoiseLGSVL , @martins-mozeiko  What does navigate and save this to assetbundles means in the following screenshot - 
[image]

asifsid-32, 2020-09-30:
> @EricBoiseLGSVL , @martins-mozeiko What does navigate and save this to assetbundles means in the following screenshot -
Which file do we have to save in this file path ?
> [image]

EricBoiseLGSVL, 2020-09-30:
That means in WebUI, when you are choosing the path for the map, save and paste to the map setup.

EricBoiseLGSVL, 2020-09-30:
The file created is fine, that is what it should have.  Linux and windows textures/assets and a manifest.  Point WebUI at this file.  Why is the file name environment_CubeTown_Files?  It should be the same as the repo/asset name

asifsid-32, 2020-09-30:
> The file created is fine, that is what it should have. Linux and windows textures/assets and a manifest. Point WebUI at this file. Why is the file name environment_CubeTown_Files? It should be the same as the repo/asset name

@EricBoiseLGSVL there was a zip file in assetbundle/environments/environment_CubeTown  which i extracted and showed you including linux and windows textures/assets and a manifest

asifsid-32, 2020-09-30:
> That means in WebUI, when you are choosing the path for the map, save and paste to the map setup.

@EricBoiseLGSVL Sorry i am not aware of this step ....

asifsid-32, 2020-09-30:
@EricBoiseLGSVL This is the zip file which is getting generated after the build inside the assetbundle/enviroments/ 
[image]

asifsid-32, 2020-09-30:
> @EricBoiseLGSVL This is the zip file which is getting generated after the build inside the assetbundle/enviroments/
> [image]

@EricBoiseLGSVL I solved the issue... We have to provide this zip file's path while adding new map 

Thanks for all the help :)

EricBoiseLGSVL, 2020-09-30:
@asifsid-32 Excellent, glad to help :)

closed 2020-10-06 by EricBoiseLGSVL