lgsvl/simulator #1035: Docker error when following ROS2 tutorial

MaxandreOgeret, 2020-09-30:
Hello everyone and thanks for reading this issue.

I am following [this](https://www.lgsvlsimulator.com/docs/autoware-auto-instructions/) tutorial for Autoware Auto with  LGSVL.

First of all the tutorial needs updating as `.aderc-nvidia` has been replaced with `.aderc-lgsvl`.

I followed the tutorial exactly and I get the following error when I try to start the docker container with the nvidia setup :

```bash
cd ~/adehome/AutowareAuto
source .aderc-nvidia
ade start
```

```
ade start                  
Starting ade with the following images:
ade          | 27321a2759b0 | master | registry.gitlab.com/autowarefoundation/autoware.auto/autowareauto/ade:master
autowareauto | 3c04644c0043 | master | registry.gitlab.com/autowarefoundation/autoware.auto/autowareauto:master    
cuda         | n/a          | latest | nvidia/cuda:latest                                                          
ade_registry.gitlab.com_autowarefoundation_autoware.auto_autowareauto_master
ade_nvidia_cuda_latest
docker: Error response from daemon: OCI runtime create failed: container_linux.go:349: starting container process caused "process_linux.go:449: container init caused \"process_linux.go:432: running prestart hook 0 caused \\\"error running hook: exit status 1, stdout: , stderr: nvidia-container-cli: mount error: file creation failed: /var/lib/docker/overlay2/8bdbcaccb623980a54b0d9c75bbcaa464bbb452d38a81877ea2f487a75a31f7e/merged/dev/nvidia-uvm: input/output error\\\\n\\\"\"": unknown.
ERROR: Command return non-zero exit code (see above): 125
  docker run -h ade --detach --name ade --env COLORFGBG --env EMAIL --env GIT_AUTHOR_EMAIL --env GIT_AUTHOR_NAME --env GIT_COMMITTER_EMAIL --env GIT_COMMITTER_NAME --env SSH_AUTH_SOCK --env TERM --env TIMEZONE=Europe/Tallinn --env USER=maxandre --env GROUP=maxandre --env USER_ID=1000 --env GROUP_ID=1000 -v /dev/shm:/dev/shm -v /tmp/.X11-unix:/tmp/.X11-unix -v /home/maxandre/adehome:/home/maxandre --env ADE_CLI_VERSION=4.2.0 --env ADE_HOME_HOSTPATH=/home/maxandre/adehome --label ade_version=4.2.0 --env DISPLAY --env VIDEO_GROUP_ID=44 -v /dev/dri:/dev/dri -v /home/maxandre/.ssh:/home/maxandre/.ssh -v /run/user/1000/keyring/ssh:/run/user/1000/keyring/ssh --volumes-from ade_registry.gitlab.com_autowarefoundation_autoware.auto_autowareauto_master:ro --volumes-from ade_nvidia_cuda_latest:ro --label 'ade_volumes_from=["ade_registry.gitlab.com_autowarefoundation_autoware.auto_autowareauto_master", "ade_nvidia_cuda_latest"]' --cap-add=SYS_PTRACE --net=host --privileged --add-host ade:127.0.0.1 -e RMW_IMPLEMENTATION=rmw_cyclonedds_cpp --runtime=nvidia -ti --rm -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -e NVIDIA_VISIBLE_DEVICES=all -e NVIDIA_DRIVER_CAPABILITIES=compute,utility,display --env ADE_IMAGE_ADE_FQN=registry.gitlab.com/autowarefoundation/autoware.auto/autowareauto/ade:master --env ADE_IMAGE_ADE_COMMIT_SHA=27321a2759b05a02e5e21bce2cf3984e710a35a7 --env ADE_IMAGE_ADE_COMMIT_TAG= --env ADE_IMAGE_AUTOWAREAUTO_FQN=registry.gitlab.com/autowarefoundation/autoware.auto/autowareauto:master --env ADE_IMAGE_AUTOWAREAUTO_COMMIT_SHA=3c04644c004361e6ed144e8a0ebe67d1b714d77d --env ADE_IMAGE_AUTOWAREAUTO_COMMIT_TAG= --env ADE_IMAGE_CUDA_FQN=nvidia/cuda:latest --env ADE_IMAGE_CUDA_COMMIT_SHA= --env ADE_IMAGE_CUDA_COMMIT_TAG= --label 'ade_images=[{"fqn": "registry.gitlab.com/autowarefoundation/autoware.auto/autowareauto/ade:master", "commit_sha": "27321a2759b05a02e5e21bce2cf3984e710a35a7", "commit_tag": ""}, {"fqn": "registry.gitlab.com/autowarefoundation/autoware.auto/autowareauto:master", "commit_sha": "3c04644c004361e6ed144e8a0ebe67d1b714d77d", "commit_tag": ""}, {"fqn": "nvidia/cuda:latest", "commit_sha": "", "commit_tag": ""}]' registry.gitlab.com/autowarefoundation/autoware.auto/autowareauto/ade:master

```

Here is the nvidia-smi on the host machine : 
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 450.51.06    Driver Version: 450.51.06    CUDA Version: 11.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce GTX 105...  On   | 00000000:01:00.0 Off |                  N/A |
| N/A   58C    P3    N/A /  N/A |   1246MiB /  4042MiB |     12%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

``` 

Do you have any idea where this error could come from ? Thanks a lot !

EricBoiseLGSVL, 2020-09-30:
@hadiTab any idea on this issue?

hadiTab, 2020-09-30:
@MaxandreOgeret The issue is related to the nvidia docker toolkit or nvidia driver. There's a discussion about it [here](https://github.com/NVIDIA/nvidia-docker/issues/1225). There don't seem to be many good solutions there. Your driver seems up-to-date.

Just to confirm, you do have nvidia-docker installed, right?
I kind of remember having this issue in the past, and it going away by reinstalling my nvidia drivers. But I may have also installed a newer version.

MaxandreOgeret, 2020-09-30:
Thanks for your answer.
Yes unfortunately there doesn't seem to be any good solution anywhere. I will try to reinstall my nvidia driver and see if anything change. Thanks a lot

endland, 2020-12-10:
How about restart docker "$ sudo systemctl restart docker"