lgsvl/simulator #1036: [unity] editor crash while opening the project

XuehuanYang, 2020-10-01:
Hi,

I've install Unity 2019.3.15f1 version on a new Ubuntu 1604 desktop. 

I'd like to build the simulator of release 2020.06.

However, when I tried to open the project with editor, the editor crashes. 

Below is the log i got:

Player.log
```
Mono path[0] = '/data/xuehuan/lgsvlsimulator-linux64-2020.06/simulator_Data/Managed'
Mono config path = '/data/xuehuan/lgsvlsimulator-linux64-2020.06/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Display 0 'HP Z23n 23"': 1920x1080 (primary device).
Loading player data from /data/xuehuan/lgsvlsimulator-linux64-2020.06/simulator_Data/data.unity3d
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Vulkan detection: 2
Initialize engine version: 2019.3.15f1 (59ff3e03856d)
[Subsystems] Discovering subsystems at path /data/xuehuan/lgsvlsimulator-linux64-2020.06/simulator_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] Graphics queue count=1
[Vulkan init] extensions: count=100
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_depth_stencil_resolve, enabled=0
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=1
[Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHR_draw_indirect_count, enabled=0
[... 865 lines cut ...]

```
Thu Oct  1 14:54:16 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.64       Driver Version: 440.64       CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce RTX 208...  Off  | 00000000:04:00.0  On |                  N/A |
| 30%   31C    P8    22W / 250W |    315MiB / 11016MiB |      2%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1622      G   /usr/lib/xorg/Xorg                           157MiB |
|    0      3736      G   /opt/teamviewer/tv_bin/TeamViewer              6MiB |
|    0      4010      G   compiz                                        75MiB |
|    0      4541      G   ...quest-channel-token=9699004864049869327    32MiB |
|    0      5304      G   ...AAAAAAAAAAAACAAAAAAAAAA= --shared-files    39MiB |
+-----------------------------------------------------------------------------+

```

hadiTab, 2020-10-01:
Seems like it's Vulkan related. I believe only `libvulkan1` is needed for the simulator. Do you need `libvulkan-dev` for something else? If not can you try removing it and opening the editor again with only `libvulkan1` installed.

closed 2020-10-20 by EricBoiseLGSVL

vijeshkumarraju, 2021-06-04:
@XuehuanYang : Did you manage to fix this error? would you please share the solution?

XuehuanYang, 2021-06-05:
I reinstalled the software and built system. Finally, the issue solved. Thx