lgsvl/simulator #1037: I can't open map data with unity

kotakky, 2020-10-01:
Hello,LGSVL team.

I'm amateur for LGSVL and  have some question for LGSVL's map.
I want to open map in  LGSVL simulator  and editing HD map format on unity.
I read the  tutorial, but I can't.
I finished LGSVL's install and unity install until now.
please tell me what I should do in detail.

kotakky, 2020-10-02:
@luqiang21 
Thank you for reply.
In Unity, Simulation tab doesn't exit. And I can't understand where the downloded default map file move to.

My screen is follwing
[image]

luqiang21, 2020-10-02:
@kotakky Are you using the correct version of Unity? Latest we are using 2019.3.15f1. The downloaded environment assets should be placed in `simulator/Assets/External/Environments`. You can refer here https://www.lgsvlsimulator.com/docs/assets/#adding-an-asset

kotakky, 2020-10-02:
I downloaded  unity of latest version 2019.3.15f1.But, I cannnot find simulator directory.
so I should make simulator directory or Asset Bundle. Is it correct to refer to https://www.lgsvlsimulator.com/docs/build-instructions/?

lemketron, 2020-10-02:
> Is it correct to refer to https://www.lgsvlsimulator.com/docs/build-instructions/?

@kotakky Yes, those are the latest build instructions, and from your screen shot it doesn't appear that you have checked out or built the simulator source code. You will need to clone the simulator source code from GitHub in order to obtain the Map Annotation functionality.

There's a video tutorial on [YouTube](https://youtu.be/EsCVYI6mhL4) that explains the entire process for downloading and building the simulator. The build instructions and video were written for a previous version of the simulator before we upgraded to Unity 2019.3.15f1, but aside from making sure you have the correct Unity version, the docs and/or video should help ensure that you're not missing any important steps. 

After that, you can refer to @luqiang21's comment [above](https://github.com/lgsvl/simulator/issues/1037#issuecomment-702523995) regarding placing the environment assets and [here](https://github.com/lgsvl/simulator/issues/1037#issuecomment-702301702) for accessing and using the annotation tool.

kotakky, 2020-10-04:
WebUI dosen't work well.I cannnot understand what happend in cmd.
I appreciate it if you could tell me what to do.

[image]

kotakky, 2020-10-05:
Thank you for your help!!
I can do "bulid" in unity and make "Build" folder that contains the unity data. 
But, I don't know about registration of the map and vehicle data.I cannot register by using local PC's path. 
From now on, I will make my own map data. I want to register the map data in lgsvl simulator web page.
I want to know about the registration method.

EricBoiseLGSVL, 2020-10-20:
@kotakky You need to create a build of your new map and it will be created in the simulator/AssetBundles folder.  Then open WebUI and select maps.  Create a name and link the path to your new asset bundle.  Boom, you have registered your new map!

closed 2020-10-20 by EricBoiseLGSVL