lgsvl/simulator #1039: Fish_eye Camera Issue

asifsid-32, 2020-10-01:
Hii LGSVL Team,

I am still facing the issue with the brightness of fish_eye camera on my ego vehicle. What can we do to improve this ?
[image]

EricBoiseLGSVL, 2020-10-05:
@asifsid-32 You will need to set a fixed exposure in HDRP settings until we can work on this issue.  Since exposure is calculated async, it varies for camera rotation and gives different values.  We plan on a fix for this soon but editing HDRP will fix this now.  It will be a little dark at night.

asifsid-32, 2020-10-12:
> @asifsid-32 You will need to set a fixed exposure in HDRP settings until we can work on this issue.  Since exposure is calculated async, it varies for camera rotation and gives different values.  We plan on a fix for this soon but editing HDRP will fix this now.  It will be a little dark at night.  See #913 

Hii @EricBoiseLGSVL is there anyway I can get the image from my fisheye camera sensor directly from the unity editor as I get from save_camera_image.py ??

EricBoiseLGSVL, 2020-10-12:
@asifsid-32 Yes, we are posting the camera image to the visualizer, please look at this code in the UIManager and sensors to see how to access the Unity RenderTexture.

asifsid-32, 2020-11-09:
> @asifsid-32 You will need to set a fixed exposure in HDRP settings until we can work on this issue. Since exposure is calculated async, it varies for camera rotation and gives different values. We [...]

Hii @EricBoiseLGSVL as you said in https://github.com/lgsvl/simulator/issues/945 
"In the meantime you can set Exposure to fixed and 0 in PostProcessingVolume.prefab to eliminate the issue. It's in Assets/Prefabs/"

I am not able to find Exposure in PostProcessingVolume.prefab. Is it somewhere else that i have to look? Can you help with this ?

EricBoiseLGSVL, 2020-11-09:
Type in PostProcessingVolume in the Unity editor project panel.  It is in Assets/Prefabs.  The volume is Exposure.  It needs set to Fixed and 0.

asifsid-32, 2020-11-09:
> Type in PostProcessingVolume in the Unity editor project panel.  It is in Assets/Prefabs.  The volume is Exposure.  It needs set to Fixed and 0.

Hii @EricBoiseLGSVL so if i change this in unity will the changes get reflected for the cameras which are there in json script ?

EricBoiseLGSVL, 2020-11-09:
Yes, this is a global post process volume

asifsid-32, 2020-11-10:
> Yes, this is a global post process volume

Thanks @EricBoiseLGSVL it helped :)