lgsvl/simulator #1040: Getting "Ground Truth Obstacles"(2D and 3D Bounding Boxes)  from lgsvl latest version 2020.06

Avps1, 2020-10-04:
Hi,
I want to get the Ground Truth Obstacles(2D and 3D Bounding Boxes)  from lgsvl for the lgsvl version 2020.06 
I had already gone through the following page : 
https://www.lgsvlsimulator.com/docs/perception-ground-truth/
But this is not updated for the newest version that I am using. Could you kindly tell me how to get the Bounding Boxes from lgsvl ?
I also looked into https://www.lgsvlsimulator.com/docs/ground-truth-json-example/#json-configuration and could not find any sensor names like 3D Ground Truth and 2D Ground Truth under sensor name .

lemketron, 2020-10-07:
You should be able to add the "2D Ground Truth" and "3D Ground Truth" sensors to your vehicle sensor json configuration as described in the link you included above.  The sensor name has not changed, and you can publish it on whatever topic you wish for your needs. 

What sensor bridge are you you using and how are you trying to read the published information?

We recently published a variation of this feature called "[Modular Testing](https://www.lgsvlsimulator.com/docs/modular-testing/)", using the same ground truth sensor names. This is intended to replace the output of perception and traffic light modules in Apollo so we publish the information to different channels (as documented in the link above). It works quite well; it is also possible to monitor the ground truth output in `cyber_monitor`, as shown here:
[image]

Avps1, 2020-10-13:
Thank you for your response. I want to create a ros listener to subscribe to the channel /simulator/ground_truth/3d_detection and /simulator/ground_truth/2d_detection.

Avps1, 2020-10-17:
I am working in Apollo 5.0, trying to build a cyber listener for the channel /simulator/ground_truth/3d_detection - I looked up the predefined protobuf msgs for Cyberrt here : https://www.lgsvlsimulator.com/docs/simulator-messages/#cyberrt-message-types
But could not find a predefined msg type for the channel channel /simulator/ground_truth/3d_detection  - Am I missing something ?

ntutangyun, 2020-10-18:
Maybe [this](https://github.com/lgsvl/lgsvl_msgs) is what you're looking for

closed 2020-10-20 by EricBoiseLGSVL