lgsvl/simulator #1041: Cant get data when using multiple Lidars

momala, 2020-10-05:
Hi,
I am using 4 lidars in our simulated vehicle. Also for merging them I use the Autoware processor feature "points_concat_filter". But it seems there is no point cloud that was published in ROS side. I also have some message in the info bar in LGSVL.
[image]

and here is my lidar definitions:
`{"type": "VelodyneLidar","name": "VLP-16_Aft",
"params": {"MinDistance": 0.10,"MaxDistance": 100,"RotationFrequency": 10,"MeasurementsPerRotation": 360,"Compensated": true,"CenterAngle": 0,"PointColor": "#ff000000","PointSize": 3,"Topic": "/points_raw1","Frame": "velodyne","VelodyneLidarType": "VLP_16"},
"transform": {"x": 0,"y": 2.25,"z": 0,"pitch": -10,"yaw": 0,"roll": 0}},{"type": "VelodyneLidar","name": "VLP-16_Fore",
"params": {"MinDistance": 0.10,"MaxDistance": 100,"RotationFrequency": 10,"MeasurementsPerRotation": 360,"Compensated": true,"CenterAngle": 0,"PointColor": "#0800ff","PointSize": 3,"Topic": "/points_raw2","Frame": "velodyne","VelodyneLidarType": "VLP_16"},
"transform": {"x": 0,"y": 2.25,"z": 2.6,"pitch": 10,"yaw": 0,"roll": 0}},{"type": "VelodyneLidar","name": "Lidar_left",
"params": {"MinDistance": 0.20,"MaxDistance": 100,"RotationFrequency": 10,"MeasurementsPerRotation": 360,"Compensated": true,"CenterAngle": -45,"FieldOfView": 85,"PointColor": "#29ff1a","PointSize": 3,"Topic": "/points_raw3","Frame": "velodyne","VelodyneLidarType": "VLP_16"},
"transform": {"x": -0.7,"y": 1.5,"z": 2.455,"pitch": 0,"yaw": 0,"roll": 90}},{"type": "VelodyneLidar","name": "Lidar_right",
"params": {"MinDistance": 0.20,"MaxDistance": 100,"RotationFrequency": 10,"MeasurementsPerRotation": 360,"Compensated": true,"CenterAngle": -45,"FieldOfView": 85,"PointColor": "#ffff2e","PointSize": 3,"Topic": "/points_raw4","Frame": "velodyne","VelodyneLidarType": "VLP_16"},
"transform": {"x": 0.7,"y": 1.5,"z": 2.455,"pitch": 0,"yaw": 0,"roll": -90}}`

There should be sth wrong that I did not consider! Thanks for your help

hadiTab, 2020-10-05:
@momala the `VelodyneLidar` sensor does not publish ROS messges and instead publishes packets over UDP that are similar to the packets a real VelodyneLidar would send. To use it, you would need to run a Velodyne ROS driver on the ROS side and configure it to listen to your lidars and publish the correct ROS message.

I believe you want to use the `Lidar` sensor that is a generic configurable lidar and publishes a pointcloud message in ROS.

momala, 2020-10-06:
> @momala the `VelodyneLidar` sensor does not publish ROS messges and instead publishes packets over UDP that are similar to the packets a real VelodyneLidar would send. To use it, you would need to [...]

Yes, Thank you for your useful information. I changed the "type" to "Lidar" and now it publishes to the ros topic. here is my setting:
`    {
        "type": "Lidar",
        "name": "VLP_Fore",
        "params": {
            "LaserCount": 16,
            "FieldOfView": 30,
            "CenterAngle": 0,
            "MinDistance": 0.10,
            "MaxDistance": 100,
            "RotationFrequency": 10,
            "MeasurementsPerRotation": 360,
            "Compensated": true,
            "PointColor": "#0800ff",
            "PointSize": 3,
            "Topic": "/points_fore", 
            "Frame": "velodyne"
        },`

closed 2020-10-06 by EricBoiseLGSVL