lgsvl/simulator #1043: How to control the variable Reverse ?

Mojojojooo, 2020-10-07:
Hey, My path has a forward and reverse section. I was to initiate reverse gear. Can I know how can I do that?

ntutangyun, 2020-10-07:
use `Page Up` and `Page Down` to toggle forward and reverse gear.

Mojojojooo, 2020-10-07:
@ntutangyun Sorry, I forgot to mention. I want to do it using ROS and not by the keyboard.

Mojojojooo, 2020-10-09:
@hadiTab I want to initiate a reverse maneuver so how can I control that variable?

EricBoiseLGSVL, 2020-10-12:
@Mojojojooo You will need to add logic to the VehicleControlSensor.cs or create your own for different ROS control.  Use this sensor to get the Dynamics class and when the subscribed msg is received, change dynamics.reverse to the new value.

Mojojojooo, 2020-10-13:
Hey, still have some doubts please help me out:
1. In VehicleControlSensor there are 4 controller options. Which one is being used when we are using lgsvl with ROS? ( autowareAI, autowareAuto, Apollo or lgsvl )
2. Code:
  controlType = ControlType.AutowareAi;
                    if (data.ShiftGearUp || data.ShiftGearDown)
                    {
                        if (data.ShiftGearUp) Dynamics.GearboxShiftUp();
                        if (data.ShiftGearDown) Dynamics.GearboxShiftDown();

                        ADAccelInput = data.Acceleration.GetValueOrDefault() - data.Breaking.GetValueOrDefault(); // converted from lin accel 
                        ADSteerInput = data.SteerAngle.GetValueOrDefault(); // angle should be in degrees
                    }
                    else
                    {
                        if (Dynamics.Reverse) return; // TODO move?
Is this the part of the code which needs to be modified so that it subscribes to the data and the Reverse status is updated?
3. If the above part is where the changes need to be made can anyone help me with adding it as I have no experience in C# as well as Unity?

Thanks in advance.

closed 2020-10-20 by EricBoiseLGSVL