lgsvl/simulator #1044: Is it possible to run lgsvl on windows and connect with Autoware on VM?

InitialDuan, 2020-10-07:
I want to run lgsvl on win10 and communicate with Autoware on VMware workstation, is that possible?
If possible, what kind of configurations needs to be done?

EricBoiseLGSVL, 2020-10-12:
@InitialDuan We have not done this but we have multiple issues answered on running simulator on VM.  It should be possible for Autoware and as long as you have open communication to LGSVL Simulator specified.  Do you have any specific issues when trying to do this?

lemketron, 2020-10-12:
Instead of running Autoware in a VM on Windows, you might want to check out this article about [Setting up ROS in Windows through WSL2](https://jack-kawell.com/2020/06/12/ros-wsl2/). 

That article references an [older one (for WSL)](https://jack-kawell.com/2019/06/24/ros-wsl1/) where the author writes:

> I’ve often wanted a way to develop within ROS on top of Windows. This can be accomplished by spinning up a VM within VirtualBox or VMWare and simply diving into a full Ubuntu desktop, but that takes up precious resources and also means dealing with all the miscellaneous issues with a full Linux install that I’m trying to avoid. Put simply, I wish for a lightweight ROS development environment which will allow me to quickly jump in and out of it without slowing down my Windows host. There are two ways to accomplish this: one way is to simply load up a Docker image based on Ubuntu with ROS pre-installed and away you go. However, Docker containers don’t have a great connection with either your network or your local files and thus aren’t tightly integrated with the Windows platform they are hosted by. The second way to accomplish this is by using WSL (Windows Subsystem for Linux)

This sounds like it might be a better approach than a full VM. Let us know what you find out!

closed 2021-02-18 by lemketron