lgsvl/simulator #1045: Do you know why my vehicle_cmd topic only returns zeros ??

Amina88, 2020-10-07:
I have installed ROS melodic and autoware.ai (1.12) i can the topic being published and subscribed when using `rostopic  list -v`
Is it normal to only display zeros like below:  My aim to get something sensible from command below and to be late able to control the vehicle.

`rostopic echo /vehicle_cmd
header: 
  seq: 3
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
steer_cmd: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  steer: 0
accel_cmd: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  accel: 0
brake_cmd: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  brake: 0
lamp_cmd: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  l: 0
  r: 0
gear: 0
mode: 0
twist_cmd: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  twist: 
    linear: 
      x: 0.0
      y: 0.0
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: -0.0
ctrl_cmd: 
  linear_velocity: 0.0
  linear_acceleration: 0.0
  steering_angle: 0.0
emergency: 0
---`

hadiTab, 2020-10-07:
@Amina88 the `twist_cmd` portion of the message is the only part that actually does anything. Giving an x value in the linear part will drive it forward and a z value in the rotational part will rotated the wheels. It's got some calibration issues that we haven't addressed yet but it can drive reasonably.

Amina88, 2020-10-09:
The twist_cmd still doesn't do anything either :( Do you have a list of cmd of how you did it as i don't seen to get anything sensible

hadiTab, 2020-10-09:
@Amina88 can you list the steps you are going through?
Are you trying to use Autoware.AI to control the vehicle or are you just trying to manually publish vehicle_cmd to control it with your own code?

Amina88, 2020-10-10:
I'm trying to use autoware.ai to control the vehicle. in terms of steps i installed ros melodic and autoware.ai and running the rosbag demo.

hadiTab, 2020-10-16:
The rosbag probably doesn't include the vehicle control message.
Overall, you can't really run the simulator with the rosbag publishing data. You can follow the steps in this guide to run the simulator with Autoware.ai: https://www.lgsvlsimulator.com/docs/autoware-instructions/

I would also suggest using version 1.14 of Autoware instead if possible.

closed 2020-10-20 by EricBoiseLGSVL