lgsvl/simulator #1047: Calibration details for fisheye cameras used in simulator

asifsid-32, 2020-10-08:
Hello LGSVL team, 

I am trying to calibrate the fisheye camera used in simulator. Now as provided we have DistortionParameters": [-0.25349, 0.11868, 0, 0]". Can anyone provide the camera matrix K which was used during calibrating the fisheye camera. Also gamma1, gamma2 values are also required. Since the fisheye model is different(Unified projection model) it is difficult for us to find the camera matrix.

Thanks in advance.

rongguodong, 2020-10-08:
[image]
I have added a FAW about camera matrix. We will push it to our website soon. But here is a screenshot for your reference.

If you are not using unified projection model, you need to study this model to figure out how to make it matching the model you are using. Please refer to the following paper and an open-source implementation for more details:
http://www.robots.ox.ac.uk/~cmei/articles/single_viewpoint_calib_mei_07.pdf
https://github.com/hengli/camodocal

For "DistortionParameters":
In most of articles of camera calibration, you may find an array as:
[image]
Here, k_1, k_2 and k_3 are the first three numbers in our "DistortionParameters" (we use up to forth order, so we have a forth number there). We ignored tangential distortion, so we do not use p_1 and p_2.

asifsid-32, 2020-10-08:
> [image]
> I have added a FAW about camera matrix. We will push it to our website soon. But here is a screenshot for your reference.
>
> If you are not using unified projection model, you need to study [...]

@rongguodong ,
If we have a fisheye image of size 1280X720 and the FOV is 50 what will be the focal length for such case ? 
If that is not possible it will be better if you can directly give us the camera matrix (K) for the unified projection model, that will be a great help.

We are not able to correct the fisheye image properly from whatever camera matrix we are using.

rongguodong, 2020-10-08:
720/2/tan(50/2/180*Pi) = 772.
This would be the value of f_1 and f_2 in formula (1) in the paper. So the matrix K in that formula will be:
[772   0   640
   0   722 360
   0     0     1]

asifsid-32, 2020-10-09:
@rongguodong What is the horizontal FOV used in the fisheye camera sensor ? Since the only FieldOfView given is 50 and that is vertical as you mentioned. Need horizontal FOV .. ??