lgsvl/simulator #1049: Question about 3D Ground Truth sensor

atinfinity, 2020-10-08:
I hope to add `3D Ground Truth` sensor to collect data for 3D object detection(LiDAR).
In this case, should the `transform` of `3D Ground Truth` sensor be the same as `transform` of Lidar sensor?
https://www.lgsvlsimulator.com/docs/sensor-json-options/#3d-ground-truth

And, I hope to subscribe to ground truth ROS messages and visualize 3D bounding box on RViz.
Is there a way to visualize 3D bounding box?
https://www.lgsvlsimulator.com/docs/perception-ground-truth/

hadiTab, 2020-10-08:
This is up to the user. If you set the transform to be the same as the Lidar sensor, the relative position of the 3D bounding boxes will be computed relative to the position the lidar is at.

As for visualizing in rviz @daviduhm might have a better idea.

atinfinity, 2020-10-11:
@hadiTab Thank you for your reply.

> This is up to the user. If you set the transform to be the same as the Lidar sensor, the relative position of the 3D bounding boxes will be computed relative to the position the lidar is at.

I understood.

> You can try using this for visualization in rviz.

I tried it. But, it seems that other car is buried in the ground.

[image]

And, the direction of the car is wrong.

[image]

atinfinity, 2020-10-12:
I change sample script.

```python
#!/usr/bin/env python

import rospy
from geometry_msgs.msg import Point
from geometry_msgs.msg import Quaternion
from geometry_msgs.msg import Vector3
from jsk_recognition_msgs.msg import BoundingBox, BoundingBoxArray
from lgsvl_msgs.msg import Detection3D, Detection3DArray, BoundingBox3D
import tf.transformations

relayed_box_array = BoundingBoxArray()

def talker():
    global relayed_box_array
    pub = rospy.Publisher('relayed_bbox', BoundingBoxArray, queue_size=10)
    num=len(relayed_box_array.boxes)
    if num>0:
        msg_str = "showing bbox %s" % num
        rospy.loginfo(msg_str)
        pub.publish(relayed_box_array)

def relay(detection_array):
    global relayed_box_array
    relayed_box_array.header.frame_id = 'velodyne'
    relayed_box_array.header.stamp = detection_array.header.stamp
    for det in detection_array.detections:
        relayed_box = BoundingBox()
        relayed_box.header.frame_id = 'velodyne'
        relayed_box.header.stamp = detection_array.header.stamp
        angles = tf.transformations.euler_from_quaternion([det.bbox.position.orientation.x, det.bbox.position.orientation.y, det.bbox.position.orientation.z, det.bbox.position.orientation.w])
        ori = tf.transformations.quaternion_from_euler(angles[0], angles[1], angles[2])
        baselink2velodyne_offset = 0.94
        relayed_box.pose.position = Point(det.bbox.position.position.x, det.bbox.position.position.y, det.bbox.position.position.z + baselink2velodyne_offset)
        relayed_box.pose.orientation = Quaternion(ori[0], ori[1], ori[2], ori[3])
        relayed_box.dimensions = det.bbox.size
        relayed_box.label = 0
        relayed_box_array.boxes.append(relayed_box)
    talker()
    relayed_box_array.boxes=[]

def listener():
    rospy.init_node('bbox_relay', anonymous=True)
    rospy.Subscriber('simulator/ground_truth/3d_detections', Detection3DArray, relay)
    rospy.spin()

if __name__ == '__main__':
    try:
        listener()
    except rospy.ROSInterruptException:
        pass
```

[image]

atinfinity, 2020-10-15:
@hadiTab If you possible, could you give me your comment?

EricBoiseLGSVL, 2020-10-20:
@hadiTab Is our sample script wrong or is this an issue with the lidar relative to the ground truth?